RedApparat/Fotoapparat #236: Better way to load Bitmap without rotation issue

SAGARSURI, 2018-04-01:
I am using your same kotlin code. I saw you were rotating the imageView to get the correct orientation of the image. Below is your code from the example:
```
val imageView = findViewById<ImageView>(R.id.result)
imageView.setImageBitmap(it.bitmap)
imageView.rotation = (-it.rotationDegrees).toFloat()
```
This works fine if the ImageView is small in the layout. But if the imageView is covering the screen width and height, the issue is that the imageView is not covering the whole screen width and height due to rotation. My suggestion would be to rotate Bitmap instead of imageView. Below is my code which I always use and gives me perfect orientation:
```
val imageView = findViewById<ImageView>(R.id.result)
myBitmap = rotateBitmap(it.bitmap, -270)
 imageView.setImageBitmap(myBitmap)
```
Method `rotateBitmap(bitmap,degree)`:
```
private fun rotateBitmap(bitmap: Bitmap, rotationAngleDegree: Int): Bitmap {
        val w = bitmap.width
        val h = bitmap.height

        var newW = w
        var newH = h
        if (rotationAngleDegree == 90 || rotationAngleDegree == 270) {
            newW = h
            newH = w
        }
        val rotatedBitmap = Bitmap.createBitmap(newW, newH, bitmap.config)
        val canvas = Canvas(rotatedBitmap)

        val rect = Rect(0, 0, newW, newH)
        val matrix = Matrix()
        val px = rect.exactCenterX()
        val py = rect.exactCenterY()
        matrix.postTranslate((-bitmap.width / 2).toFloat(), (-bitmap.height / 2).toFloat())
        matrix.postRotate(rotationAngleDegree.toFloat())
        matrix.postTranslate(px, py)
        canvas.drawBitmap(bitmap, matrix, Paint(Paint.ANTI_ALIAS_FLAG or Paint.DITHER_FLAG or Paint.FILTER_BITMAP_FLAG))
        matrix.reset()

        return rotatedBitmap
    }
```

SAGARSURI, 2018-04-07:
@dmitry-zaitsev  What is your suggestion on this ?

dmitry-zaitsev, 2018-04-17:
@SAGARSURI sorry for the delayed answer.

Rotating the bitmap is, of course, solving the problem. However, it is a slow and memory intensive operation as you would essentially need to load the copy of the original image into memory.

I will take that problem into account - I think we might address it in Fotoapparat. Here is the related ticket: https://github.com/Fotoapparat/Fotoapparat/issues/241

closed 2018-04-17 by dmitry-zaitsev

JcmeLs, 2018-09-07:
@SAGARSURI hello,I have same issue.Do you have any update?