RedApparat/Fotoapparat #243: Why is bitmap null?

mqwangios, 2018-04-18:
Why is bitmap null?  

private class SampleFrameProcessor implements FrameProcessor {
        @Override
        public void process(@NotNull Frame frame) {
            // Perform frame processing, if needed
            Bitmap bitmap = BitmapFactory.decodeByteArray(frame.getImage(), 0, frame.getImage().length);
            ImageView imageView = findViewById(R.id.result);
            imageView.setImageBitmap(bitmap);
        }

dmitry-zaitsev, 2018-05-16:
Frames of the `FrameProcessor` are in NV21 format which can't be decoded using `BitmapFactory`. You can still convert it to `Bitmap` but you would have to use another approach: https://stackoverflow.com/questions/32276522/convert-nv21-byte-array-into-bitmap-readable-format

closed 2018-05-16 by dmitry-zaitsev

mqwangios, 2018-05-20:
How to set the output size, I think the output is 1080*1920?
[image]

bagage, 2018-06-19:
For anyone interested, most straightforward method is to use [easyrs](https://github.com/silvaren/easyrs) library:

```gradle
//in build.gradle
implementations 'io.github.silvaren:easyrs:0.5.3'
```

Then convert NV21 to bitmap: 

```java
RenderScript rs = RenderScript.create(context); // where context can be your activity, application, etc.
Bitmap outputBitmap = Nv21Image.nv21ToBitmap(rs, nv21ByteArray, width, height); // where nv21ByteArray contains the NV21 image data
```