RedApparat/Fotoapparat #244: Java documentation could be easier

JamesMcIntosh, 2018-04-18:
The example on the [Configuration-Java](https://github.com/Fotoapparat/Fotoapparat/wiki/Configuration-Java) page would be a lot better if it included either the class names which contain the constants / helper methods. i.e.

```
Fotoapparat fotoapperat = Fotoapparat
    .with(getContext())
    .into(cameraView)                             // view which will draw the camera preview
    .previewScaleType(ScaleType.CenterCrop)       // we want the preview to fill the view  
    .photoSize(highestResolution())               // we want to have the biggest photo possible
    .lensPosition(LensPositionSelectorsKt.back())       // we want back camera
    .focusMode(SelectorsKt.firstAvailable(  // (optional) use the first focus mode which is supported by device
            FocusModeSelectorsKt.continuousFocusPicture(),
            FocusModeSelectorsKt.autoFocus(),        // in case if continuous focus is not available on device, auto focus will be used
            FocusModeSelectorsKt.fixed()             // if even auto focus is not available - fixed focus mode will be used
    ))
    .flash(SelectorsKt.firstAvailable(      // (optional) similar to how it is done for focus mode, this time for flash
            FlashSelectorsKt.autoRedEye(),
            FlashSelectorsKt.autoFlash(),
            FlashSelectorsKt.torch()
    ))
    .frameProcessor(myFrameProcessor)   // (optional) receives each frame from preview stream
    .logger(LoggersKt.loggers(            // (optional) we want to log camera events in 2 places at once
            LoggersKt.logcat(),           // ... in logcat
            LoggersKt.fileLogger(this)    // ... and to file
    ))
    .build();
```
Note: I didn't find highestResolution()

siralam, 2018-04-27:
Extremely agree, I struggled for a long time on not being able to import the static functions like `continuousFocus`, `autoFocus`, `fixed`. `continuousFocus` and `photoSize` have been renamed but not updated in the documentation.

Diolor, 2018-07-04:
Since several people had issues with that I updated the docs :)

closed 2018-07-04 by Diolor