RedApparat/Fotoapparat #248: Camera Error 100

tapeo, 2018-04-26:
Hi! The camera never starts with the following error:

```
04-26 17:38:10.120 7366-7385/com.test E/Camera: New camera parameters are: CameraParameters
    flashMode:		io.fotoapparat.parameter.Flash$Off@42023fe8
    focusMode:		io.fotoapparat.parameter.FocusMode$Fixed@420112c8
    jpegQuality:		90
    previewFpsRange:		FpsRange(min=3000, max=30000)
    antiBandingMode:		io.fotoapparat.parameter.AntiBandingMode$None@41ff6bc0
    sensorSensitivity:		null
    pictureResolution:		Resolution(width=1600, height=1200)
    previewResolution:		Resolution(width=800, height=600)
04-26 17:38:10.500 7366-7379/com.test W/CameraBase: Camera service died!
04-26 17:38:10.500 7366-7380/com.test W/CameraBase: mediaserver's remote binder Camera object died
04-26 17:38:10.500 7366-7366/com.test E/Camera: Error 100
04-26 17:38:10.520 7366-7385/com.test I/ServiceManager: Waiting for service media.camera...
```

Initialized with:

```
camera = Fotoapparat.with(getActivity())
                .into(preview)
                .lensPosition(front())
                .previewScaleType(ScaleType.CenterCrop)
                .focusMode(firstAvailable(
                        fixed(),
                        infinity(),
                        continuousFocusPicture(),
                        autoFocus()
                ))
                .logger(new Logger() {
                    @Override
                    public void log(String s) {
                        Log.e("Camera", s);
                    }

                    @Override
                    public void recordMethod() {

                    }
                })
                .cameraErrorCallback(new CameraErrorListener() {
                    @Override
                    public void onError(CameraException e) {
                        e.printStackTrace();
                    }
                })
                .build();
```

Maybe a SurfaceHolder error? Thank you!

Diolor, 2018-04-27:
It means you have an app using camera service already and still not released the camera. Could also be that you have two Fotoapparats active

closed 2018-04-27 by Diolor

tapeo, 2018-04-27:
Two devices with no external apps installed, in one device works all perfect, in the other not works. Maybe not compatible hardware? I investigate it now.

tapeo, 2018-04-27:
I found the fix. Removing `antiBandingMode` from parameters in `CameraParametersApplicator` the camera starts without errors. Consider to fix this if you feel necessary.