RedApparat/Fotoapparat #251: Is there a "FitStart" Scale Type?

siralam, 2018-04-27:
As far as I know there are only 2 scale types that can be configured in `previewScaleType()`, known as `ScaleType.CenterCrop` and `ScaleType.CenterInside`.

Now suppose I don't want my preview to be cropped (because I need to overlay something on it on a exact position). So I must use `CenterInside`.

The problem is, no matter I set `match_parent` or `wrap_content` to `CameraView`'s xml, it always fill the screen, and the preview is centered in the screen. And that leaves blank area in both top and bottom space; but **I need my preview to be pinned at the top**.

Is this possible?

Diolor, 2018-04-27:
Why not use center crop with wrap content in height?

siralam, 2018-04-27:
The preview fills the screen if I use `CenterCrop` with `wrap_content` in height.

DubyOriginal, 2018-05-10:
... I have the same problem