RedApparat/Fotoapparat #252: FrameProcessor returns frame has low FPS

siralam, 2018-05-02:
By looking at the log, `process()` of `FrameProcessor` is called only 3-4 times every second; while I tried to implement my own camera (still using Camera1 now), I can get up to 27 FPS.

Is this a limitation of the library or Camera 2 API?

enricoros, 2018-05-30:
With the following code I get 27-29fps. Also depends on where the camera is looking at - if dark/black, it goes down to 5 fps or so (normal).

    private ArrayList<Long> mLastFramesTimes = new ArrayList<>();
    private int mAvgFps;

    private FrameProcessor myFrameProcessor = new FrameProcessor() {
        @Override
        public void process(Frame frame) {
            // FPS counter (uses a 10frames trailing window to compute)
            long thisTime = System.currentTimeMillis();
            mLastFramesTimes.add(thisTime);
            if (mLastFramesTimes.size() <= 20)
                return;
            long firstTime = mLastFramesTimes.remove(0);
            float avgTime = (float) (thisTime - firstTime) / (float) (mLastFramesTimes.size() - 1);
            mAvgFps = (int) Math.round(1000. / Math.max(1, avgTime));
        }
    };

Diolor, 2018-07-04:
Camera v1 will give you a "good" (30fps) frame rate. Camera v2 will give a horrible 1fps. 
For this reason we only use the legacy android camera v1 here.

closed 2018-07-04 by Diolor

siralam, 2018-07-04:
Sad to hear this reply. I turned out implemented my own own CameraView and I got good FPS with Camera2.

Diolor, 2018-07-04:
That's interesting. Did you use ImageReader or something else to get the frames?

siralam, 2018-07-04:
I used `ImageReader`, specifying `YUV_420_888` as the image format. This makes the FPS from Camera high.

But user has to decode by themselves from YUV_420_888 to ARBG so that Android can use it.  
For this, I personally used RenderScript.

You may take a look at my [Camera2 class](https://github.com/siralam/CameraViewPlus/blob/master/cameraviewplus/src/main/api21/com/google/android/cameraview/Camera2.java)

And I used [this library](https://github.com/silvaren/easyrs) for converting YUV_420_888 to ARGB.

siralam, 2018-07-04:
You're welcomed, hope fotoappart will grow better and better!