RedApparat/Fotoapparat #255: Camera "shape correction" on Samsung Galaxy S7 defaults to enabled.

jeroenlicht, 2018-05-07:
The Samsung Galaxy S7 has a "shape correction" feature that distorts the image in the middle and is messing up the accuracy of our photo measurements. This is how it affects the image: [VIdeo showing the distortion](https://www.youtube.com/watch?v=jlDhGsUrBhc)

Comparing the photos from our app using Fotoapparat with the photos made with the Samsung camera app shows a similar distortion. The Samsung Camera app has "shape correction" turned off by default, but it seems that Fotoapparat enables it, and I can see no way of controlling it.

These are the capabilities and parameters I get from the S7 using the Fotoapparat library in our app:

`Capabilities{photoSizes=[Size{width=3024, height=3024}, Size{width=2160, height=2160}, Size{width=2048, height=1152}, Size{width=4032, height=3024}, Size{width=2880, height=2160}, Size{width=2560, height=1440}, Size{width=2976, height=2976}, Size{width=3264, height=2448}, Size{width=4032, height=2268}, Size{width=640, height=480}, Size{width=3264, height=1836}, Size{width=3984, height=2988}, Size{width=960, height=720}, Size{width=1920, height=1080}, Size{width=1280, height=720}], previewSizes=[Size{width=1056, height=704}, Size{width=176, height=144}, Size{width=352, height=288}, Size{width=800, height=450}, Size{width=1024, height=768}, Size{width=720, height=720}, Size{width=1088, height=1088}, Size{width=256, height=144}, Size{width=720, height=480}, Size{width=1440, height=1080}, Size{width=640, height=480}, Size{width=320, height=240}, Size{width=1920, height=1080}, Size{width=960, height=720}, Size{width=1280, height=720}], focusModes=[CONTINUOUS_FOCUS, INFINITY, FIXED, MACRO, AUTO], flashModes=[TORCH, ON, AUTO, OFF], previewFpsRanges=[[15000, 15000], [15000, 30000], [30000, 30000], [24000, 24000]], supportedSensorSensitivityRange={100, 200, 400, 800}, zoomSupported=true}`

`Parameters{values={FOCUS_MODE=CONTINUOUS_FOCUS, FLASH=ON, PICTURE_SIZE=Size{width=960, height=720}, PREVIEW_SIZE=Size{width=1440, height=1080}}}`

Diolor, 2018-05-07:
Android camera api has no such functionality. Therefore all camera SDKs will experience such distortion by default. Looks like Samsung introduced this feature in their camera app as a fix of their faulty hardware. https://i.imgur.com/YYGOzUk.png

If you feel like fixing this we would be more than happy to accept a PR.

jeroenlicht, 2018-05-07:
> as a fix of their faulty hardware
Good point. It seems like the back camera of the S7 has significant lens distortion, with or without shape correction. It's just distorted one way or the other. 
Perhaps the front camera provides better geometry than the back camera.

Diolor, 2018-07-04:
I will close this one unless you are willing to dive into distortion corrections and fix it :)

closed 2018-07-04 by Diolor