RedApparat/Fotoapparat #259: Camera not released even after stop() is called

chittorakalpit, 2018-05-30:
I used fotoapparat to handle the camera. After the event was completed, I called ```fotoapparat.stop()``` {fotoapparat is an instance of Fototapparat class}

And next step was to open camera for another activity which cannot be done. 

Alert Message pops:
"Seems there is an issue with the camera. This could be because of other apps running on phone or a hardware issue. Restarting this device might solve this problem."

dmitry-zaitsev, 2018-05-30:
When are you calling `stop()`? Could you perhaps share some code with us? That would help.

chittorakalpit, 2018-05-31:
I'm using the FrameProcessor for live processing. Then when result is found I call ```fotoapparat.stop()```. What I want is that the code should wait for the camera to be released.

dmitry-zaitsev, 2018-05-31:
The `FrameProcessor` is working in its own thread which means that it very well might be that the frame is processed already _after_ Activity is stopped. You would need to handle that case.

On top of that, `Fotoapparat` should always be operated from UI thread. Therefore make sure you are not calling `stop()` directly from `FrameProcessor` but first at least wrap it in `runOnUiThread` (or use `Handler`, `View.post` or whatever else to make it run on UI thread).

closed 2018-05-31 by dmitry-zaitsev