RedApparat/Fotoapparat #262: Incorrect camera view orientation when device turned upside down

dlevache-pro, 2018-06-06:
When the screen orientation is not locked, the system does not create a new activity when the device is turned upside down (from portrait to reverse portrait  or from landscape  to reverse landscape, with no intermediate orientation).
In both these cases, the camera view displays images in the wrong orientation.

With an intermediate orientation (portrait during the landscape to reverse landscape switch), there is no problem.

#### What are you trying to achieve or the steps to reproduce?
Using the code from the sample:
* Start the application
* The camera view is rendered correctly
[image]
* Turn the device upside down.

#### What was the result you received?
The camera view is rendered upside down.
[image]

#### What did you expect?
Video preview is rendered in the correct orientation

#### Context:

* FA version: 2.2.0 / master
* Devices affected: Samsung XCover3 (5.1.1) Nexus 6P (8.1)

dmitry-zaitsev, 2018-06-08:
Thanks for reporting that. I will take a look at it.

marekfoltyn, 2018-07-03:
The same bug appears in Samsung Galaxy S5. The last thing required to fix in order to have perfect camera view 📸 :)

marekfoltyn, 2018-09-07:
Update: Since the android does not recreating Activity when switching landscape modes, it has to be done manually. I fixed this issue by manual detecting Display orientation. When landscape <-> reversed_landscape change is detected the following code is called:
```
runOnUiThread {
    fotoapparat.stop()
    fotoapparat.start()
}
```

ijwhelan, 2018-12-21:
Hey @marekfoltyn, I'm running into this same issue when going from landscape -> reverse landscape. How are you going about manually detecting display orientation?

rdsarna, 2019-02-11:
Hi @ijwhelan, if you're targeting API 17+ then the following fix should work -
Include this file in your project -
[https://gist.github.com/rdsarna/4b2bb5315fcad9356011cf37d9073826](url)

Then, in your onCreate() add the following code - 
```
if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.JELLY_BEAN_MR1) {
    reverseOrientationObserver = ReverseOrientationObserver(this) {
        fotoapparat.stop()
        fotoapparat.start()
    }
}
```

And in onDestroy() add - 
```
reverseOrientationObserver?.quit()
reverseOrientationObserver = null
```

balazsbanto, 2019-02-14:
Hi @rdsarna 
Your link does not work. Could you fix it please? I also encountered this problem. Thanks in advance$

rdsarna, 2019-02-14:
@balazsbanto Sorry about that, fixed now.

commit 1783058a (2019-03-02): Fixes orientation issues (#354)

* fix: preview orientation in sensorLandscape

* Fix orientation issues.

Fixes #262
Fixes #260
Fixes #253
Fixes #193
closed 2019-03-02 by Diolor