RedApparat/Fotoapparat #264: toSingle() or pendingResult.await() cause camera to freeze

Raenar4k, 2018-06-19:
#### What are you trying to achieve or the steps to reproduce?

I am trying to use fotoapparat Rx adapters to work with the result or get result directly from pendingResult.await() method.
Both are causing camera to freeze.

I've reproduced this issue in sample project, using version from `master` branch or from `2.2.0` tag.
You can use either method and take a photo from back camera to reproduce this.

Here im using await() method:
```kotlin
    private fun takePicture(): () -> Unit = {
        val photo = fotoapparat
                .autoFocus()
                .takePicture()
                .toBitmap(scaled(scaleFactor = 0.25f))
                .await()
        
        Log.i(LOGGING_TAG, "New photo captured. Bitmap length: ${photo.bitmap.byteCount}")

        val imageView = findViewById<ImageView>(R.id.result)

        imageView.setImageBitmap(photo.bitmap)
        imageView.rotation = (-photo.rotationDegrees).toFloat()
    }
```

And here is the example same as in fotoapparat-adapters readme, using rxjava2: 
```groovy
dependencies {
    implementation project(':fotoapparat')

    implementation project(':fotoapparat-adapters:rxjava2')
    implementation "io.reactivex.rxjava2:rxjava:2.1.12"

    implementation "org.jetbrains.kotlin:kotlin-stdlib:${versions.kotlin}"

    implementation "com.android.support:appcompat-v7:${versions.android.support}"
}
```

```kotlin
    private fun takePicture(): () -> Unit = {
        fotoapparat.takePicture()
                .toBitmap()
                .toObservable()
                .subscribe { photo ->
                    Log.i(LOGGING_TAG, "New photo captured. Bitmap length: ${photo.bitmap.byteCount}")

                    val imageView = findViewById<ImageView>(R.id.result)

                    imageView.setImageBitmap(photo.bitmap)
                    imageView.rotation = (-photo.rotationDegrees).toFloat()
                }
    }
```
Using `toFlowable()` or `toObservable()` produces same result.

#### How did you initialize FA?
Sample project looks like this (shortened):
```kotlin
        fotoapparat = Fotoapparat(
                context = this,
                view = cameraView,
                focusView = focusView,
                logger = logcat(),
                lensPosition = back(),
                cameraConfiguration = CameraConfiguration(
                        previewResolution = firstAvailable(
                                wideRatio(highestResolution()),
                                standardRatio(highestResolution())
                        ),
                        previewFpsRange = highestFps(),
                        flashMode = off(),
                        focusMode = firstAvailable(
                                continuousFocusPicture(),
                                autoFocus()
                        ),
                        frameProcessor = {
                            // Do something with the preview frame
                        }
                ),
                cameraErrorCallback = { Log.e(LOGGING_TAG, "Camera error: ", it) }
        )
```
#### Context:

* *FA version*: 2.2.0
* *Devices/APIs affected*: tested on Google pixel, Samsung s9
* *any other relevant information*:
In logs the last entry is "CameraDevice: takePhoto" - then device freezes.
```
06-19 16:28:19.943 8246-8246/io.fotoapparat.sample D/Fotoapparat: Fotoapparat: autoFocus
    Fotoapparat: focus
06-19 16:28:19.945 8246-8246/io.fotoapparat.sample D/Fotoapparat: Fotoapparat: takePicture
06-19 16:28:19.953 8246-8264/io.fotoapparat.sample D/Fotoapparat: CameraDevice: autoFocus
06-19 16:28:22.962 8246-8264/io.fotoapparat.sample D/Fotoapparat: CameraDevice: takePhoto
```

Diolor, 2018-06-20:
Docs from `await`: `Blocks current thread until result is available.`. So please use it in a thread.

Regarding `toObservable` looks like you also block main thread too and you call `takePicture` from that (main) thread.

So you need to use different threads:
```
.subscribeOn(Schedulers.io())
.observeOn(AndroidSchedulers.mainThread())
```

closed 2018-06-20 by Diolor

Raenar4k, 2018-06-21:
Ah, my bad! Thanks for quick response. 
Yep, it works alright, when triggered in non UI thread.

I did expect it to block, but return some result eventually.  
As to why callback never gets delivered:
`onPictureTaken()` callback is called on main thread. `takePicture` method is also executed on main thread and it seems to block thread until callback is called -> callback can not be called since the thread is blocked by `takePicture` method -> and so it is never returned. 

While it is asynchronous (`"Triggers an asynchronous image capture"`), it does not execute on some background thread, it executes on same thread it was called.

Hence, if you expect it block for a while and then deliver result, you are wrong as i am :)

Is everything correct, @Diolor ?

Diolor, 2018-06-21:
- If you call `.await()` the task will be executed and result will be reported on same thread.
- If you call `whenAvailable` the task will be executed on a background thread but result will be reported on main thread.
- If you use Rx Adapters (without manual thread changes) the task will be executed and result will be on same thread because rxjava is by default synchronous.

When main thread is blocked because it's waiting the result no other operation can be executed e.g. taking picture itself internally.

Raenar4k, 2018-06-22:
Now it makes sense to me!
Thanks :)