RedApparat/Fotoapparat #266: Camera preview is blackout.

monowar1993, 2018-06-21:
Camera preview is full blackout and logcat is prtinting
> E/Camera: Error 1
Camera was working fine in on 7.1.1 with my nexus 5. Also it is working with Xiaomi Mi A1 8.0.0. But I updated my Nexus to 8.1.0. From then its not working with my nexus 5.

#### What are you trying to achieve or the steps to reproduce?
I tested the sample app. Sample app is also throwing the same error.

#### Context:

* *FA version*: 2.2.0
* *Devices/APIs affected*: Nexus 5, 8.1.0 (LineageOS 15.1), Rooted

dmitry-zaitsev, 2018-06-21:
Can you confirm that Camera permission is granted?

monowar1993, 2018-06-21:
Yes. I did. Just confirm me its working with 8.1.0 for now.

dmitry-zaitsev, 2018-06-21:
Then I suppose we can close this issue until the problem reappears :)

closed 2018-06-21 by dmitry-zaitsev

monowar1993, 2018-06-21:
I asked is this library working on other 8.1.0 devices?? If yes. Then its good to go.

monowar1993, 2018-06-21:
Ok. Thanks. :)

joshuaherman, 2018-07-31:
Essential PH-1 Release 8.1.0 - library works.