RedApparat/Fotoapparat #277: Difference between preview ratio and photo ratio

Twister21, 2018-07-20:
Hello,

I've noticed that images created by Fotoapparat don't fill the full smartphone screen, so that you can see black bars on the image's top and bottom. However, there are no black bars on the preview screen, which means that the screen is filled completely. To do so, the status bar is hidden.

How can I configure Fotoapparat to take photos with the ScaleType CenterCrop which is already set for the preview image by `previewScaleType(ScaleType.CenterCrop)`

Diolor, 2018-07-20:
Firstly, the end picture has no extra black content. I assume you are using an ImageView to display your photos and the black area is the absence of photo drawing in that area. Because simply the photo ration is different than the screen.

Assuming those you need to check your ImageView's attributes, especially the [scale type](https://developer.android.com/reference/android/widget/ImageView.ScaleType). But this is a general Android topic and has nothing to do with this library. :)

Twister21, 2018-07-20:
I think that you don't understand the main issue. The photo taken by Fotoapparat should be in a full screen mode like the photos which are taken by the original Android Camera app. However, photos taken by the Android camera app and photos taken by Fotoapparat differ in the described aspect of black bars, which you can see with the Android gallery app. The image solution is exactly the same, but not the aspect ratio.

Twister21, 2018-07-21:
@Diolor Can you please open the issue again?

Diolor, 2018-07-21:
Okay let's go back to square one.

I am talking here about the captured picture, **not** the preview:

_The image resolution is exactly the same, but the aspect ratio not._ : This is not possible, if you have two pictures of 4000 x 3000 then the ratio is also the same: 4:3.

Are you talking that end picture is 4000 x 3000 but the "real" picture is e.g. 3000 x 3000 and another 1000 x 3000 is black area? If yes, in which devices can you replicate this?

Twister21, 2018-07-21:
Oh, sorry it's my mistake. The photo taken by the Android camera app has a resolution of 3264x1836, whereas Fotoapparat takes photos with a resolution of 3264x2448. So the image ratio is different and you can see black bars on the photos taken by Fotoappart. But why are there used two different sizes for the second value? The image ratio should be 16:9.

Diolor, 2018-07-21:
The status quo:
Most hardware camera sensors are supporting 4:3 at max resolution even Nikon and Canon DSLR cameras. They can also do 16:9 but this is a crop of the max 4:3.
So a 3264x2448 resolution is a 8MP at 4:3 and 3264x1836 is a 6MP at 16:9. The 16:9 is a crop (subset) of the 4:3 picture.

This case:
Firstly which phone are you having this issue? I am really curious because it's the first time I know which is happening.

This phone probably is not the "most phones" case and has a sensor doing a max 16:9.
However that manufacturer tells Fotoapparat (and the Android camera API) wrongly that the device can capture a 3264x2448, which is not case since you see the two black areas (the difference between a 4:3 and a 16:9). So the sensor supports 16:9 and then if you request a 4:3, it takes the 16:9 and then fills black areas to make the rest of 4:3.

The manufacturer also knows that this phone can capture max a 3264x1836 so they can ship their native app happily with an option of max 6MP (@ 3264x1836). The weird part would have been if the native app could capture successfully a full 8MP (@ 3264x2448).

So if the manufacturer tells us that the device can do a 3264x2448, we trust them and we capture at this size. Otherwise they violate the API support meaning it's a bug on their side.

Diolor, 2018-07-22:
You can simply crop the picture if: it's this phone model and it's a 3264x2448: make it a 3264x1836.

For you it's a visible issue but the phone is quite old so will its usage. Therefore the impact of this bug is extremely low to introduce a fix to just for this phone.

Twister21, 2018-07-22:
Do you know how to detect the phone model and are there any other known devices where this error could occure.

Twister21, 2018-07-22:
So  `.photoResolution(ResolutionSelectorsKt.highestResolution())` chooses the 4:3 resolution although I just want to use 16:9 resolutions? How can I limit this descision?

Diolor, 2018-07-22:
- Do you know whether there are any other known devices where this error could occure?
- No

- Do you have any explanation why Fotoapparat takes the wrong resolution? 
- There is no right or wrong resolution. I have already explained above that FA takes what the device provides. 4:3 is higher than 16:9 for same height. Therefore it takes 4:3. The black areas are what the Camera api gives us.

- So .photoResolution(ResolutionSelectorsKt.highestResolution()) chooses the 4:3 resolution although I just want to use 16:9 resolutions? How can I limit this descision?
- `wideRatio(highestResolution())`

I will close this issue because we will not make a fix for this device due to low impact. This is a device/lower level camera api bug and not related to FA.

closed 2018-07-22 by Diolor

Twister21, 2018-07-22:
Which version implements wideRatio()? I'm using Fotoapparat with Java and there isn't a function called wideRatio, just photoResolution.