RedApparat/Fotoapparat #279: Unable to create Bitmap from preview Frame

ClaudiuBele, 2018-07-25:
#### What are you trying to achieve or the steps to reproduce?

I want to save the last preview `Frame` to a file instead of the result of  `photoApparat.takePicture()`, and in that process i need to make a Bitmap object from the `Frame`. BitmapFactory calls return null, I've been trying to following
```
BitmapFactory.decodeByteArray(frame.image, 0, frame.size.area)
// and
BitmapFactory.decodeStream(ByteArrayInputStream(frame.image))
// and
val bitmap = Bitmap.createBitmap(frame.size.width, frame.size.height, Bitmap.Config.ARGB_8888)
val byteBuffer = ByteBuffer.wrap(frame.image)
bitmap.copyPixelsFromBuffer(byteBuffer)
```
The first 2 attempts return null, last one returns exception `Buffer not large enough for pixels`. Is there any other option to save the preview `Frame` to file or getting a bitmap for it?

Otherwise, the library works as intended

#### Context:

* *FA version*:2.2.0

ChristianLJ, 2018-08-06:
I have the same problem.

ChristianLJ, 2018-08-06:
I managed to fix it. I checked the source code and it seems that the image format is NV21 which BitmapFactory doesn't support.

I wrote the below code which converts the YUV image to a Bitmap.

```java
YuvImage yuvImage = new YuvImage(frame.image, ImageFormat.NV21, width, height, null);
ByteArrayOutputStream os = new ByteArrayOutputStream();
yuvImage.compressToJpeg(new Rect(0, 0, width, height), 100, os);
byte[] jpegByteArray = os.toByteArray();
Bitmap bitmap = BitmapFactory.decodeByteArray(jpegByteArray, 0, jpegByteArray.length);
```

dmitry-zaitsev, 2019-01-22:
@ChristianLJ that is exactly correct, frames are in NV21 format which is not supported by `BitmapDecoder`. Thanks for sharing your solution 👍

closed 2019-01-22 by dmitry-zaitsev