RedApparat/Fotoapparat #280: Setting null FrameProcessor in updateConfiguration() still starts frames processing. Some devices receive null preview frames which causes IllegalStateException.

xaeon, 2018-07-26:
When i call the Fotoapparat.updateConfiguration() function, i set frameProcessor = null (the default value). But inside the library, the PreviewStream.updateProcessorSafely() function is still called with an empty, but not null, FrameProcessor. So this function starts frame processing. There is no problems other than increased resources consumption. 

But on the ZTE BLADE A520 device, PreviewCallback.onPreviewFrame() get called with null data parameter. When it get passed to PreviewStream.dispatchFrameOnBackgroundThread function, a  "java.lang.IllegalStateException: data must not be null" is thrown

wilsonrc, 2018-08-06:
@xaeon I think I have the same Issue with a Samsung Galaxy Grand Prime with Android 7.1, do you already have any solution for this?

wilsonrc, 2018-08-12:
@dmitry-zaitsev can you help us with this?

dmitry-zaitsev, 2018-08-13:
Thanks for reporting it. It should be fixed by this PR:

https://github.com/RedApparat/Fotoapparat/pull/284

You can expect a new version in 2 days.

closed 2018-08-13 by dmitry-zaitsev

dmitry-zaitsev, 2018-08-13:
The fix is available in version 2.3.3.