RedApparat/Fotoapparat #295: Can't start preview because of the exception: java.io.IOException: setPreviewTexture failed

marekfoltyn, 2018-09-10:
Sometimes (about 1 of 15 attempts) preview won't start, because of the following exception:

`D/Fotoapparat: Can't start preview because of the exception: java.io.IOException: setPreviewTexture failed`

It happens usually when the device is being rotated (the Activity gets recreated) on both my devices (Samsung Galaxy S5 and Samsung Galaxy Tab S2).

fb89, 2018-10-01:
I think, it happens because Texture may not be available yet (it became destroyed once you rotate your device). So `onSurfaceTextureAvailable/onSurfaceTextureDestroyed` methods must be used for saving actual Texture state here `io.fotoapparat.view.CameraView`

marekfoltyn, 2018-10-08:
Thanks for your advice. If I understand it correctly, you think that I should initialize `Fotoapparat` instance in `cameraView.onSurfaceTextureAvailable` method?