RedApparat/Fotoapparat #296: How to get maxZoom in Java?

scifinder, 2018-09-10:
Hello! I'm using Java. I do not understand how to get the value of maxZoom. `Capabilities.getZoom()` and then?

marekfoltyn, 2018-09-14:
Hi, check the method documentation:
```
   /**
     * Asynchronously updates zoom level of the camera.
     * If zoom is not supported by the device - does nothing.
     *
     * @param zoomLevel Zoom level of the camera. A value between 0 and 1.
     * @throws IllegalStateException If the current camera has not started.
     */
    fun setZoom(...) ...
```

I think the zoom in Fotoapparat is relative - 0 is no zoom and 1 is max zoom. In my app I just use 5 zoom steps (0, 0.25, 0.5, 0.75, 1).

johntzan, 2018-09-21:
This is how I did it with Java: 

    private void getZoomCapability(){

        fotoapparat.getCapabilities().whenDone(capabilities ->  {
                if(capabilities==null){
                    return;
                }
                setupZoom((Zoom.VariableZoom) capabilities.getZoom());
            }
            
        );
        
    }
    

and then in `setupZoom(Zoom.VariableZoom zoom)` I am getting max Zoom with: `zoom.getMaxZoom()`. Be sure to call this after fotoapparat has started.

Hope that helps!

scifinder, 2018-10-23:
@johntzan Thank you very much! You helped me a lot.

scifinder, 2018-10-23:
But... On some devices I get a value of 30 or 100.

closed 2019-03-02 by Diolor