RedApparat/Fotoapparat #303: Custom aspect ratio is not working

Nominalista, 2018-09-30:
I wanted to set the aspect ratio for the preview and picture the same as screen aspect ratio.

```
val displayMetrics = DisplayMetrics()
activity.windowManager.defaultDisplay.getMetrics(displayMetrics)
val screenAspectRatio = displayMetrics.heightPixels.toFloat() / displayMetrics.widthPixels.toFloat()

val cameraConfiguration = CameraConfiguration(
        pictureResolution = aspectRatio(screenAspectRation, highestResolution()),
        previewResolution = aspectRatio(screenAspectRation, highestResolution()))
```
Unfortunately as the result I get black screen (despite the fact that `screenAspectRatio` is correct value). 4:3 and 16:9 ratios are working correctly.

dmitry-zaitsev, 2018-10-10:
What happens is that Fotoapparat tries to select a size with an aspect ratio which is exactly equal to `screenAspectRatio` (you have a typo btw). It is very unlikely that there is a size with exactly matching ratio so nothing is selected and you see a black screen.

`aspectRatio()` has an optional 3rd parameter which defines a tolerance of how far the actual aspect ratio might be from what you defined.

closed 2018-10-10 by dmitry-zaitsev

Nominalista, 2018-10-24:
Thanks, I assumed that Fotoapparat can choose any aspect ratio.