RedApparat/Fotoapparat #304: Custom photoResolution

Corical, 2018-10-05:
Not an issue but question on how to get a custom resolution, something along the lines of 800x600, we don't want highest resolution and lowest is too low, something in between otherwise the pictures take up a lot of space and we're definitely not sending 4mb a pop (highest reso).

Should I take them at high reso and scale them down manually ?

Diolor, 2018-10-06:
It's not possible. Simply if a phone camera doesn't support e.g. 800x600, you can't have it. You can always take a bigger one and downscale it by yourself.

closed 2018-10-06 by Diolor

Gnzlt, 2018-11-30:
How do I know if the phone camera doesn't support `800x600` beforehand?

Diolor, 2018-12-08:
@Gnzlt 
https://github.com/RedApparat/Fotoapparat/blob/master/fotoapparat/src/main/java/io/fotoapparat/Fotoapparat.kt#L128