RedApparat/Fotoapparat #311: io.fotoapparat.exception.camera.CameraException: Failed to open camera with lens position: io.fotoapparat.characteristic.LensPosition$Back@31c3b8a2 and id: 0

devangichhatbar, 2018-10-21:
Please check if your issue exists.

Issues should only be posted after you have been able to reproduce them and confirm that they are either a missing functionality or a bug. 

#### What are you trying to achieve or the steps to reproduce?

Describe your issue here, include as much detail as neccessary to reproduce the issue
or implement a missing functionality.

```java
// Wrap code in markdown source tags
```

#### How did you initialize FA?

```java
// Wrap code in markdown source tags
```

#### What was the result you received?

#### What did you expect?

#### Context:

* *FA version*:
* *Devices/APIs affected*:
* *any other relevant information*:

Pareshoct7, 2018-10-22:
Same issue.

wellbranding, 2018-10-23:
Same issue. I have added  runnable with delay 200 ms, when starting fotoappart camera. I am really sad that this library does not get any improvements...

selmantsn, 2018-12-03:
Same issue. Is there any solution?

dineshvg, 2018-12-21:
I made a workaround where we just restart the fragment or activity that holds the camera preview on the occurance of this error. It is shabby, but the user does not see a camera-view with an empty preview when the screen is opened after minimization.

DaanGeurts, 2019-02-04:
I had the same issue did you ask and granted the camera permission? I resolved it by granting the app camera access.

ForestTree, 2019-02-07:
Same problem, but ONLY on Huawei P9 Lite (Android 7.0). On the other tested devices, the problem does not occur (Huawei P8 Lite, Moto G4, Pixel 2). As mention @wellbranding , short delay before calling `start()` solved this issue.

DPalagi, 2019-02-13:
Same issue here, only for activities where I force the requestedOrientation to landscape at start.

Even with the 200ms delay, it's not working...

wellbranding, 2019-03-14:
Same issue in 2.7.0. Should we stop fotoappart in onDestroyView or in onStop in the Fragment? What are your suggestions to avoid this problem?

Diolor, 2019-03-15:
hey @wellbranding, do you still experience blank/black screens? Those should have been fixed in 2.7.0 due to a threading issue that we had.

To the lifecycle topic, well...there is no rule of the thumb. Depends your activities and how they work. Usually, start/stop is ok. Others can use resume/pause but camera will be paused in e.g. in split mode. I would avoid onDestroy because the activity GC is not guaranteed.

wellbranding, 2019-03-15:
@Diolor blank/blacks screens seems to be fixed, indeed :) It works well with onStart and onStop with Activity. The problem occurs with Fragments. 

Supppose I have a fragment, which will start an Activity, which has cameraSession (regular, not from fotoappart).

 After onStop in that activity I experience same error here. We are still testing if it is our issue or not :) 

Nevertheless, thanks for your recommendations! We will continue to use Fotoappart in our SDK, which gets implemented across apps :)

Diolor, 2019-03-15:
That's nice :) So, the hack of the 200ms should not be needed anymore. Give it a try and remove it, I would suggest.

Maybe some stop() operation is not executed somehow/sometime because of the handler or some other issue.

I would close issue this since blank screens are now gone 🎉

wellbranding, 2019-03-15:
@Diolor yes, it seems to work well now! I create instance of Fotopparat inside of **onActivityCreated** in the **fragment** and then use onStop and onStart to handle stop() and start() accordingly.

wellbranding, 2019-03-15:
@Diolor  sadly, but issue continues.. Our testers team found that if you quickly open camera, then go to background, remove fragment and then will try to  open fragment again you will encounter this error:
Can't start preview because of the exception: java.io.IOException: setPreviewTexture failed
which will then create a blank screen and will cause sameio.fotoapparat.characteristic.LensPosition$Back@31c3b8a2 and id: 0 #311

Seems not right :(

Diolor, 2019-03-15:
A log would be a great help! 👍

wellbranding, 2019-03-15:
2019-03-15 20:15:07.308 14211-14368/com.idenfy.app.new D/Fotoapparat: CameraDevice: getPreviewResolution
2019-03-15 20:15:07.310 14211-14368/com.idenfy.app.new D/Fotoapparat: Preview resolution is: Resolution(width=1536, height=2048)
2019-03-15 20:15:07.311 14211-14368/com.idenfy.app.new D/Fotoapparat: CameraDevice: setDisplaySurface
2019-03-15 20:15:07.312 14211-14368/com.idenfy.app.new D/Fotoapparat: CameraDevice: startPreview
2019-03-15 20:15:08.279 14211-14211/com.idenfy.app.new D/Fotoapparat: Fotoapparat: stop
2019-03-15 20:15:08.283 14211-14368/com.idenfy.app.new D/Fotoapparat: CameraDevice: stopPreview
2019-03-15 20:15:08.324 14211-14368/com.idenfy.app.new D/Fotoapparat: CameraDevice: close
2019-03-15 20:15:09.268 14211-14211/com.idenfy.app.new D/Fotoapparat: Fotoapparat: start
2019-03-15 20:15:09.268 14211-14368/com.idenfy.app.new D/Fotoapparat: Device: selectCamera
2019-03-15 20:15:09.269 14211-14368/com.idenfy.app.new D/Fotoapparat: CameraDevice: open
2019-03-15 20:15:09.272 14211-14211/com.idenfy.app.new D/Fotoapparat: Fotoapparat: stop
2019-03-15 20:15:09.451 14211-14368/com.idenfy.app.new D/Fotoapparat: CameraDevice: getCapabilities$suspendImpl
2019-03-15 20:15:09.452 14211-14368/com.idenfy.app.new D/Fotoapparat: CameraDevice: updateParameters$suspendImpl
2019-03-15 20:15:09.453 14211-14368/com.idenfy.app.new D/Fotoapparat: New camera parameters are: CameraParameters
    flashMode:		Flash.Off
    focusMode:		FocusMode.Fixed
    jpegQuality:		90
    exposureCompensation:		0
    previewFpsRange:		FpsRange(min=7500, max=24000)
    antiBandingMode:		AntiBandingMode.Auto
    sensorSensitivity:		null
    pictureResolution:		Resolution(width=5184, height=3880)
    previewResolution:		Resolution(width=2048, height=1536)
2019-03-15 20:15:09.461 14211-14368/com.idenfy.app.new D/Fotoapparat: CameraDevice: updateFrameProcessor
2019-03-15 20:15:09.464 14211-14368/com.idenfy.app.new D/Fotoapparat: CameraDevice: setDisplayOrientation
2019-03-15 20:15:09.465 14211-14368/com.idenfy.app.new D/Fotoapparat: Orientations: 
    Screen orientation (preview) is: Orientation.Vertical.Portrait. 
    Camera sensor orientation is always at: Orientation.Horizontal.ReverseLandscape. 
    Camera is mirrored.
2019-03-15 20:15:09.465 14211-14368/com.idenfy.app.new D/Fotoapparat: Orientation adjustments: 
    Image orientation will be adjusted by: 90 degrees. 
    Display orientation will be adjusted by: 90 degrees. 
    Preview orientation will be adjusted by: 90 degrees.
2019-03-15 20:15:09.465 14211-14368/com.idenfy.app.new D/Fotoapparat: CameraDevice: getPreviewResolution
2019-03-15 20:15:09.468 14211-14368/com.idenfy.app.new D/Fotoapparat: Preview resolution is: Resolution(width=1536, height=2048)
2019-03-15 20:15:09.468 14211-14368/com.idenfy.app.new D/Fotoapparat: CameraDevice: setDisplaySurface
2019-03-15 20:15:09.469 14211-14368/com.idenfy.app.new D/Fotoapparat: **Can't start preview because of the exception: java.io.IOException: setPreviewTexture failed**
2019-03-15 20:15:09.471 14211-14368/com.idenfy.app.new D/Fotoapparat: CameraDevice: stopPreview
2019-03-15 20:15:09.474 14211-14368/com.idenfy.app.new D/Fotoapparat: CameraDevice: close
2019-03-15 20:15:09.837 14211-14211/com.idenfy.app.new D/Fotoapparat: Fotoapparat: getCurrentParameters
2019-03-15 20:15:10.028 14211-14211/com.idenfy.app.new D/Fotoapparat: Device: updateLensPositionSelector
2019-03-15 20:15:10.029 14211-14211/com.idenfy.app.new D/Fotoapparat: Fotoapparat: <init>
2019-03-15 20:15:10.031 14211-14211/com.idenfy.app.new D/Fotoapparat: Fotoapparat: start

In this case, error did not appeared, but we get a **blank screen.**

2019-03-15 20:19:16.088 14211-14211/com.idenfy.app.new D/Fotoapparat: Fotoapparat: stop
2019-03-15 20:19:16.089 14211-14946/com.idenfy.app.new D/Fotoapparat: Couldn't deliver pending result: Camera operation was cancelled.
2019-03-15 20:19:16.091 14211-14368/com.idenfy.app.new D/Fotoapparat: CameraDevice: stopPreview
2019-03-15 20:19:16.138 14211-14211/com.idenfy.app.new D/Fotoapparat: Device: updateLensPositionSelector
2019-03-15 20:19:16.138 14211-14368/com.idenfy.app.new D/Fotoapparat: CameraDevice: close
2019-03-15 20:19:16.139 14211-14211/com.idenfy.app.new D/Fotoapparat: Fotoapparat: <init>
2019-03-15 20:19:16.139 14211-14211/com.idenfy.app.new D/Fotoapparat: Fotoapparat: start
2019-03-15 20:19:16.197 14211-14211/com.idenfy.app.new D/Fotoapparat: Fotoapparat: stop
2019-03-15 20:19:16.198 14211-14368/com.idenfy.app.new D/Fotoapparat: Device: selectCamera
2019-03-15 20:19:16.198 14211-14368/com.idenfy.app.new D/Fotoapparat: CameraDevice: open
2019-03-15 20:19:16.315 14211-14211/com.idenfy.app.new D/Fotoapparat: Device: updateLensPositionSelector
2019-03-15 20:19:16.316 14211-14211/com.idenfy.app.new D/Fotoapparat: Fotoapparat: <init>
2019-03-15 20:19:16.317 14211-14211/com.idenfy.app.new D/Fotoapparat: Fotoapparat: start
2019-03-15 20:19:16.322 14211-14368/com.idenfy.app.new D/Fotoapparat: CameraDevice: getCapabilities$suspendImpl
2019-03-15 20:19:16.323 14211-14368/com.idenfy.app.new D/Fotoapparat: CameraDevice: updateParameters$suspendImpl
2019-03-15 20:19:16.323 14211-14368/com.idenfy.app.new D/Fotoapparat: New camera parameters are: CameraParameters
    flashMode:		Flash.Off
    focusMode:		FocusMode.Fixed
    jpegQuality:		90
    exposureCompensation:		0
    previewFpsRange:		FpsRange(min=7500, max=24000)
    antiBandingMode:		AntiBandingMode.Auto
    sensorSensitivity:		null
[... cut ...]
This time we get an exception.

Setup: Fragments, every fragment **replaces** other fragment, so onStop and onStart is called correctly

. In the parent activity of fragments inside of **onStart,** we remove some fragments (**after returning from background**)

I have looked internally Fotoapparat, because our team will also contribute to it soon, but I am not sure why error occurs.. We will try different setups and check how to possibly solve this issue. **No memory leaks occur**, it is well tested with LeakCanary.

wellbranding, 2019-03-16:
@Diolor  does our logcat indicates possible issue with Fotoapparat? We have tested various ways. In the end, our results indicate if we call fotoapparat.start() in the onResume instead of onStart() in the fragment following exception does not occur:

**Can't start preview because of the exception: java.io.IOException: setPreviewTexture failed**

It seems that library is functioning well, just it behaves differently with Fragments and activities, because of lyfecycle issues.

Diolor, 2019-03-16:
setPreviewTexture is coming from the Android camera API. The documentation is incomplete but from my experience, this might happen if you have a view which has not been released from previous FA, or you try to reuse an existing view which is not "clean for usage" or because Android has a bad day...

I _think_ this might have to do with object management with the lifecycle. E.g. first FA has not fully released camera and you instantly try to initiate a new one, or something wrong with the layout view.

wellbranding, 2019-03-16:
@Diolor  thanks for information. Yes it seems it is, because of those reasons. We will continue to use 2.7.0 version, but with **Handlers** then calling fotoapparat.start(). It passed our tests and seems to work well.

Replacing/ adding fragments with fotoappart object seems to be an issue..

arsenii-repository, 2019-04-11:
Experiencing same issue with 2.7.0 version. Any news/workarounds?

arsenii-repository, 2019-04-11:
@wellbranding Hi, may I ask you please to post working piece of code?

ErAmrit, 2019-06-17:
Getting same issue on G4 while using activity. When changing orientation black screen is occurred. Tried delay 200ms , also uses stop method before start. Please help

ErAmrit, 2019-06-17:
When changing orientation black screen is occurred. Tried delay 300ms , also uses stop method before start. Continuously happening in Portrait  if change from landscape
version: 2.7.0

abayomiAkanji, 2020-05-20:
Initialization of the object and calling `fotoapparat.start()` in the onResume method (of the fragment) seems to be the hack for me.
This implies that whenever you return to the fragment, it re-initializes the object and starts it.
This neither returns error nor does the camera goes blank.

noorashref, 2023-11-20:
Im using 2.6.1 and got this exception mainly for moto G power (2021) Any solution for fix ? io.fotoapparat.exception.camera.CameraException : Failed to start preview for camera with lens position: io.fotoapparat.characteristic.LensPosition$Back@c1b6b84 and id: 0