RedApparat/Fotoapparat #313: Why does Fotoapparat landscape CamewraView captures in portrait?

murathada, 2018-10-29:
I have implenented the Fotoapparat library to my app. To operate the whole app with obe hand, I am trying to make the CameraView in a portrait mode parents ConstraintLayout. As you can see in the following screenshot, the CameraView itself is landscape, but the parent ConstraintLayout of the whole Fragment is portrait mode, so that user can operate app with one hand.

[image]

Now I show the captured image as a preview in the next fragment.. As you can see, the captured image is portrait mode now:
[image]

```java
        fotoapparat = Fotoapparat
                .with(activity)
                .into(cameraView)
                .focusView(focusView)
                .sensorSensitivity(highestSensorSensitivity())
                .focusMode(firstAvailable(
                        FocusModeSelectorsKt.continuousFocusPicture(),
                        FocusModeSelectorsKt.autoFocus()))
                .lensPosition(back())
                .build();
```

Excerpts of my layouts:

```java
<io.fotoapparat.view.CameraView
    android:id="@+id/myapp_camera_view"
    android:layout_width="0dp"
    android:layout_height="0dp"
    android:layout_marginStart="@dimen/myapp_border_margin"
    android:layout_marginEnd="@dimen/myapp_border_margin"
    app:layout_constraintStart_toStartOf="parent"
    app:layout_constraintEnd_toEndOf="parent"
    app:layout_constraintDimensionRatio="1.778"
    app:layout_constraintBottom_toTopOf="@+id/myapp_border"
    app:layout_constraintTop_toBottomOf="@id/myapp_holder"
    app:layout_constraintVertical_bias="0.5">

    <io.fotoapparat.view.FocusView
        android:id="@+id/myapp_focus_view"
        android:layout_width="match_parent"
        android:layout_height="match_parent"
    />

</io.fotoapparat.view.CameraView>
```

Versions: Fotoapparat 2.4.0
Java SDK 8
Android Studio 3.2

What am I dong wrong?

Neronse, 2018-11-02:
You must rotate result for correct orientation. On some devices this angle is incorrect (few tablets i know) You can do it like this:
`photoResult
    .toBitmap()
    .whenAvailable { bitmapPhoto ->
            val imageView = (ImageView) findViewById(R.id.result)
 
            imageView.setImageBitmap(bitmapPhoto.bitmap)
            imageView.setRotation(-bitmapPhoto.rotationDegrees)
    }`
I took it from main page ;)

murathada, 2018-11-08:
Another way where I can take landscape photos while holding the smartphone portrait?