RedApparat/Fotoapparat #314: Cannot load Fotoapparat on Kotlin 1.3

bmachimbira, 2018-10-31:
I migrated my existing application that uses Fotoapparat for the camera. The application compiles successfully but I get the following error:

`java.lang.NoClassDefFoundError: Failed resolution of: Lkotlinx/coroutines/experimental/CompletableDeferredKt;
        at io.fotoapparat.hardware.CameraDevice.<init>(CameraDevice.kt:48)
        at io.fotoapparat.hardware.Device.<init>(Device.kt:39)
        at io.fotoapparat.hardware.Device.<init>(Device.kt:34)
        at io.fotoapparat.Fotoapparat.<init>(Fotoapparat.kt:55)
        at io.fotoapparat.Fotoapparat.<init>(Fotoapparat.kt:48)`

I think this is because the library uses the experimental coroutines module, which has been removed from Kotlin 1.3.

I am using version 2.5.0 of Fotoapparat.

geminixandroid, 2018-11-01:
Same issue

limuyang2, 2018-11-05:
Same issue

limuyang2, 2018-11-05:
coroutines error

limuyang2, 2018-11-05:
my version is `implementation 'org.jetbrains.kotlinx:kotlinx-coroutines-core:1.0.0'`

vicobz, 2018-11-05:
Same issue. Need a fix asap! :)

AshuTyagi16, 2018-11-06:
Same Issue. please fix.

dmitry-zaitsev, 2018-11-06:
I will take a look and try to release a new version today. Latest, tomorrow.

PR #315 (2018-11-06): Migrates project to AndroidX and Kotlin 1.3
Fixes #314
closed 2018-11-06 by Diolor