RedApparat/Fotoapparat #316: Could not find io.fotoapparat:fotoapparat:2.6.0.

Laces, 2018-11-07:
Please check if your issue exists.

Issues should only be posted after you have been able to reproduce them and confirm that they are either a missing functionality or a bug. 

#### What are you trying to achieve or the steps to reproduce?

Describe your issue here, include as much detail as neccessary to reproduce the issue
or implement a missing functionality.

```java
// Wrap code in markdown source tags
```

#### How did you initialize FA?

```java
// Wrap code in markdown source tags
```

#### What was the result you received?

#### What did you expect?

#### Context:

* *FA version*:
* *Devices/APIs affected*:
* *any other relevant information*:

fleficher, 2018-11-07:
Hi, seems that `2.6.0` version is not fully published yet. Meanwhile, you can use the `2.5.0`.

dmitry-zaitsev, 2018-11-07:
It was just deployed, sorry for the inconvenience.

closed 2018-11-07 by dmitry-zaitsev

vicobz, 2018-11-08:
I still face 
> Failed to resolve: io.fotoapparat.fotoapparat:library:2.6.0

Am i missing something?

dmitry-zaitsev, 2018-11-08:
@vicobz it should be `io.fotoapparat:fotoapparat:2.6.0`, not `io.fotoapparat.fotoapparat:library:2.6.0`

vicobz, 2018-11-08:
I was so missing something 😉
Thanks for your quick reply!