RedApparat/Fotoapparat #319: How to set a preview and picture resolution in java

hitec16, 2018-11-08:
This is not an issue but a question.
I want to always have a full-screen preview in the camera and the image with highest resolution of the same ratio.

#### What are you trying to achieve or the steps to reproduce?

 For example a 16:9 ratio preview should give a 16:9 ratio image always and not the highest quality 4:3 ratio image.

```java
// Wrap code in markdown source tags
```

#### How did you initialize FA?

```java
mFotoapparat = Fotoapparat
                .with(activity)
                .into(cameraView)
                .previewScaleType(ScaleType.CenterCrop)
                .photoResolution(ResolutionSelectorsKt.highestResolution())
                .lensPosition(LensPositionSelectorsKt.back())
                .frameProcessor(new FrameProcessor() {
                    @Override
                    public void process(Frame frame) {
                        Log.d(TAG, frame.toString());
                    }
                })
                .logger(loggers(
                        logcat(),
                        fileLogger(activity)
                ))
                .cameraErrorCallback(new CameraErrorListener() {
                    @Override
                    public void onError(@NotNull CameraException e) {
                        Toast.makeText(activity, e.toString(), Toast.LENGTH_LONG).show();
                    }
                })
                .build();
```

#### What was the result you received?

I am getting full-screen preview of 16:9 and the clicked image is of 4:3 ratio of highest quality.
I do not know Kotlin so can anyone help me like how can I pass a custom resolution using JAVA. For example in below code line.
    .photoResolution(ResolutionSelectorsKt.highestResolution())

#### What did you expect?
I expect to send one of the supported resolutions via JAVA as well.

#### Context:

* *FA version*:2.5.0
* *Devices/APIs affected*: For all
* *any other relevant information*:

Diolor, 2018-11-09:
Hey, did you check [our wiki for java](https://github.com/RedApparat/Fotoapparat/wiki/Configuration-Java)?

You cannot set an aspect ratio of a preview, because your phone might be 16/9 but others might be 16/10. Only if you crop or fit the preview in a view

ControlAdad, 2019-02-23:
@Diolor How can I set final photo resolution to say for example 1600x1200 and then call save to disk?

Diolor, 2019-03-02:
@ControlAdad You cannot have only one desired resolution size. Some phones might not support this size so size selection is dynamic

closed 2019-03-02 by Diolor