RedApparat/Fotoapparat #321: Round corners for CameraView ?

murathada, 2018-11-09:
I try to make round corners for the CameraView. The recorded image shall remain with edges. I just want to make the preview rounded, not the captured image.

I tried using a shape I've created before. 

```java
<?xml version="1.0" encoding="UTF-8"?>
    <shape xmlns:android="http://schemas.android.com/apk/res/android"
           android:shape="rectangle">
        <solid android:color="@android:color/transparent"/>
        <stroke android:width="16dp"
                android:color="#ffffff"/>
        <corners android:radius="24dp" />
    </shape>
```
then I just set this as CameraViews background. but this is not working.

dmitry-zaitsev, 2018-11-09:
You need to set it as a foreground, otherwise camera will cover your
background.

One simple way to do it is to put an ImageView on top of your Camera.

murathada, 2018-11-09:
Wrappen CameraView with ImageView:
`
<ImageView
        android:id="@+id/myapp_image_view_container"
        android:foreground="@drawable/rounded_shape"
        android:layout_width="match_parent"
        android:layout_height="0dp"
        app:layout_constraintDimensionRatio="1.6"
        app:layout_constraintEnd_toEndOf="parent"
        app:layout_constraintStart_toStartOf="parent"
        app:layout_constraintBottom_toTopOf="@+id/myapp_single_pic_border"
        app:layout_constraintTop_toBottomOf="@id/myapp_title_holder" >
    <io.fotoapparat.view.CameraView
            android:id="@+id/myapp_camera_view"
            android:layout_width="match_parent"
            android:layout_height="match_parent"
            android:paddingStart="@dimen/myapp_default_margin"
            android:paddingEnd="@dimen/myapp_default_margin"
            android:paddingTop="@dimen/myapp_border_margin"
            android:paddingBottom="@dimen/myapp_border_margin">
        <io.fotoapparat.view.FocusView
                android:id="@+id/myapp_focus_view"
                android:layout_width="match_parent"
                android:layout_height="match_parent"
                />
    </io.fotoapparat.view.CameraView>
</ImageView>
`
But I get an error therefore?

`
    android.view.InflateException: Binary XML file line #54: android.support.v7.widget.AppCompatImageView cannot be cast to android.view.ViewGroup
    Caused by: java.lang.ClassCastException: android.support.v7.widget.AppCompatImageView cannot be cast to android.view.ViewGroup
        at android.view.LayoutInflater.rInflate(LayoutInflater.java:875)
        at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:835)
        at android.view.LayoutInflater.rInflate(LayoutInflater.java:877)
        at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:835)
        at android.view.LayoutInflater.rInflate(LayoutInflater.java:877)
        at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:835)
        at android.view.LayoutInflater.inflate(LayoutInflater.java:515)
        at android.view.LayoutInflater.inflate(LayoutInflater.java:423)
        at com.example.test.ui.MyCameraFragment.onCreateView(MyCameraFragment.java:53)
`

murathada, 2018-11-09:
Attempt to do it programatically: 
```
cameraView.setBackgroundResource(R.drawable.rounded_shape);
```
leads to following:
[image]

murathada, 2018-11-09:
As you said.. foreground ;)
cameraView.setForeground(getResources().getDrawable(R.drawable.rounded));

closed 2018-11-09 by murathada