RedApparat/Fotoapparat #323: Samsung Galaxy 9  SMG-960F - On capture image there arer 2 freezes and flash gets crazy

kashkashio, 2018-11-15:
Using Fotoapparat latest version with Samsung's galaxy s9 with sample app provided in repository.
Reproduce is easy, get a device, run the app, try to capture and you will see it immediately.

It doesn't happen on galaxy s8 - s7 - s6 - s5... on these devices it works perfectly.
I couldn't notice any change in the resolutions or the sizes or fps or anything like that that can cause it.

I Tried to play with resolutions and configurations and nothing helps.

 The init is exact same as sample. i didn't touched.
No code to supply according the fact it's just the sample app provided without any changes.

#### What was the result you received?
Slow capturing and freezing twice and then gets release the preview.

#### What did you expect?
To capture flawlessly without noticing same as other devices do.

* *FA version*:Latest Version from git
* *Devices/APIs affected*:Samsung galaxy s9 - SMG-960F
* *any other relevant information*:Its important and i'd like to get help there for me and all other peoples that are going to try this great and awesome library !

P.S
Tried more projects to test it and the results are:
* this library worked flawless - https://github.com/googlesamples/android-Camera2Basic
* this one reproduce the same freeze - https://github.com/Jiankai-Sun/Android-Camera2-API-Example

Hope it will help to find a solution.

Thanks a lot!
If any info is missing just let me know and i will complete it,
And i really want to thank you all guys for the amazing job you did with this library tough i know how difficult the android api cameras are awful...
Thank you ! <3

Attached a video of the error 
https://streamable.com/s/zkp6j/gufczb

samuelgunawan, 2018-12-24:
I also experience same bug. Some freezes.

Fotoapparat Version: 2.6.1
Device: Google Pixel 3
OS: Pie 9.0 (SDK 28)

Thanks.

CabreraGerardo, 2019-10-15:
Expirenced similar issue with zoom on this device.
Library Version: 2.7.0