RedApparat/Fotoapparat #329: How do I set focal length / optical zoom (supported by hardware)?

huanghuan, 2018-12-10:
Please check if your issue exists.

Issues should only be posted after you have been able to reproduce them and confirm that they are either a missing functionality or a bug. 

#### What are you trying to achieve or the steps to reproduce?

Describe your issue here, include as much detail as neccessary to reproduce the issue
or implement a missing functionality.

```java
// Wrap code in markdown source tags
```

#### How did you initialize FA?

```java
// Wrap code in markdown source tags
```

#### What was the result you received?

#### What did you expect?

#### Context:

* *FA version*:
* *Devices/APIs affected*:
* *any other relevant information*:

dmitry-zaitsev, 2019-01-22:
@huanghuan sorry for the late reply. Did you try using default zoom methods from `Fotoapparat`? I had an impression that they would work for optical zoom as well, but perhaps I am wrong.

Diolor, 2019-03-02:
Closing this due to inactivity

closed 2019-03-02 by Diolor