RedApparat/Fotoapparat #332: Can't make a photo on an Acer Iconia A1-830

wolf81, 2019-01-07:
#### What are you trying to achieve or the steps to reproduce?

I'm trying to take a picture on an Acer Iconia A1-830 device. I don't own the device myself, but I make use of Microsoft AppCenter, which allows me to test on many different devices. I believe people can use a limited amount of hours for free every month, so one should be able to reproduce in AppCenter.

```java
// Wrap code in markdown source tags
```

#### How did you initialize FA?

```java
        this.camera = Fotoapparat
                .with(this)
                .into(cameraView)
                .lensPosition(LensPositionSelectorsKt.back())
                .photoResolution(ResolutionSelectorsKt.lowestResolution())
                .previewResolution(ResolutionSelectorsKt.lowestResolution())
                .previewScaleType(ScaleType.CenterCrop)
                .flash(off())
                .focusMode(SelectorsKt.firstAvailable(
                        FocusModeSelectorsKt.continuousFocusPicture(),
                        FocusModeSelectorsKt.autoFocus(),
                        FocusModeSelectorsKt.fixed(),
                        FocusModeSelectorsKt.infinity()
                ))
                .logger(LoggersKt.logcat())
                .cameraErrorCallback((exception) -> {
                    Log.e(TamperSecure.TAG, exception.toString());
                })
                .build();
```

#### What was the result you received?

The following is shown in the logs when trying to take a photo:

```
D/Fotoapparat( 5129): CameraDevice: open
D/Camera_HAL(  140): ATOM_GetCameraInfo
D/Camera_HAL(  140): ATOM_GetCameraInfo
I/AwesomePlayer(  140): setDataSource_l(URL suppressed)
W/MediaPlayer(  140): info/warning (1, 1130153408)
I/AwesomePlayer(  140): initAudioDecoder: creating OMX decoder
I/cip_VorbisDec(  140): UMC Decoder plugin created, Media Codecs version: 1.0.2098
I/AwesomePlayer(  140): setDataSource_l(URL suppressed)
W/MediaPlayer(  140): info/warning (1, 1130252528)
I/AwesomePlayer(  140): initAudioDecoder: creating OMX decoder
I/cip_VorbisDec(  140): UMC Decoder plugin created, Media Codecs version: 1.0.2098
I/AudioPolicyManagerBase(  140): APM: getOutput
I/AudioPolicyManagerBase(  140): APM: getOutput
I/AudioPolicyManagerBase(  140): APM: getOutput
I/CameraClient(  140): Opening camera 0
D/Camera_HAL(  140): ATOM_OpenCameraHardware
E/Camera_Conf(  140): ERROR no sensor driver registered in media controller!
D/Camera_ISP(  140): Kernel with separate preview device node detected
W/Camera_ISP(  140): Open device 0 with fd 35
D/hwcomposer(  392): isRGBOverlayLayer: got an RGB overlay layer
D/hwcomposer(  392): isRGBOverlayLayer: got an RGB overlay layer
I/ActivityManager(  392): Displayed com.peir.tampersecure_example/com.peir.tampersecure.CameraActivity: +205ms
D/hwcomposer(  392): isRGBOverlayLayer: got an RGB overlay layer
I/WAKELOCK_RELEASE(  392): TIMESTAMP=1401376269648, TAG=ActivityManager-Launch, TYPE=PARTIAL_WAKE_LOCK             , COUNT=0, PID=392, UID=1000, FLAGS=
D/hwcomposer(  392): isRGBOverlayLayer: got an RGB overlay layer
D/LifecycleMonitor( 5129): Lifecycle status change: com.peir.tampersecure_example.EventsActivity@210002d8 in: STOPPED
D/LifecycleMonitor( 5129): running callback: androidx.test.rule.ActivityTestRule$LifecycleCallback@20fe8878
D/LifecycleMonitor( 5129): callback completes: androidx.test.rule.ActivityTestRule$LifecycleCallback@20fe8878
D/hwcomposer(  392): isRGBOverlayLayer: got an RGB overlay layer
I/libva   (  140): VA-API version 0.34.0
I/libva   (  140): va_getDriverName() returns 0
I/libva   (  140): Trying to open /system/lib/pvr_drv_video.so
I/libva   (  140): Found init function __vaDriverInit_0_32
[... 96 lines cut ...]
I believe the issue has something to do with Flash, but I believe I configured Flash correctly (it's set to off and that seems to be the only possible config). This part of the log could (in my opinion) point to the issue:

```
...
D/Fotoapparat( 5129): Fotoapparat: takePicture
D/Fotoapparat( 5129): CameraDevice: takePhoto
D/Camera_HAL(  140): atom_enable_msg_type msg_type=0x00000100
D/Camera_HAL(  140): atom_take_picture
E/Camera_ISP(  140): Failed to get value for control (10225921) on device '35', Invalid argument
E/Camera_ISP(  140): Error getting Flash mode from the driver
W/Camera_ControlThread(  140): Unexpected state (0) to select the shooting mode
E/Camera_ControlThread(  140): Taking picture when recording is not supported!
E/Camera_ControlThread(  140): Error handling message: 7
...
```

#### What did you expect?

To be able to make a successful photo :)

#### Context:

* *FA version*: Android 4.2.2
* *Devices/APIs affected*: Acer Iconia A1-830
* *any other relevant information*:

dmitry-zaitsev, 2019-01-22:
Sorry for the late reply. 

It does indeed look like you are setting the flash correctly. However, just for the sake of experiment, could you try setting flash to some other values and see if that helps? If it does, it would give me an idea on where to start looking for the issue.

wolf81, 2019-03-29:
@dmitry-zaitsev I've been looking a bit deeper into this issue today and I believe another tablet (Kobo Arc 7HD) likely has the same issue.

I believe flash mode may not be set to *off* for both devices, instead the flash mode shouldn't exist at all or perhaps the value for flash should be set to nothing() / null. Both devices likely not have flash hardware anyway, since they're both tablets. I wanted to verify my assumptions, but I don't know how I would be able to set the value of flash to null (or nothing()), seems to be not allowed in fotoapparat.