RedApparat/Fotoapparat #333: Preview "resolution" to match native sensor "resolution"

huckw, 2019-01-09:
Not an issue, but a question...

I would like to operate on the full "resolution" sensor data in the frameProcessor() callback.  By "resolution" I do not mean the pixel count or sensor size, but instead I mean that i want the raw sensor data without decimation.  I understand that my sensor pixel count is much larger than my display pixel count...so in order to get the raw sensor data I want I can only get a subset of it that fits the number of pixels in my display. In a way it is like a digital zoom.  Is this at all possible with fotoapparat?

dmitry-zaitsev, 2019-01-10:
> so in order to get the raw sensor data I want I can only get a subset of it that fits the number of pixels in my display

That is not true. Frame size of is bound to the preview size of the camera. Preview size of the camera is not bound to the display size. In fact, we have to crop the preview image so that it fits the device screen.

To answer your problem, you can just set the biggest possible preview size and it will give you preview frames of the highest possible resolution.

dmitry-zaitsev, 2019-01-22:
Sorry for the late response.

No, unfortunately it is not possible to use full camera resolution for the preview due to the limitations of the camera API. I am not sure why exactly it is not allowed but I would make a wild guess that device is not able to produce preview frames fast enough to achieve an acceptable frame rate.

My suggestion is - try to periodically capture full resolution pictures and process those instead.

closed 2019-01-22 by dmitry-zaitsev