RedApparat/Fotoapparat #336: freeze when takephoto and convert to bitmap

RowanG1, 2019-01-23:
When I use the code:
fotoapparat?.takePicture()?.toBitmap(),

the preview freezes for about one second. Is there any way to overcome this? I am testing on a Xiaomi device.

Diolor, 2019-03-02:
Migration to Camera 2 is the only way which is currently not in the roadmap.

closed 2019-03-02 by Diolor

shedeed1, 2019-06-17:
I managed to get around this issue by using RxJava and subscribing on IO thread.

Disposable disposable = fotoapparat.takePicture()
                            .toBitmap()
                            .adapt(SingleAdapter.<BitmapPhoto>toSingle())
                            .subscribeOn(Schedulers.io())
                            .subscribe(new Consumer<BitmapPhoto>() ......