RedApparat/Fotoapparat #339: Changing previewResolution in updateConfiguration not working

PawelTypiak, 2019-01-25:
I am trying to change the resolution of the preview and picture to the highest resolution by using ```updateConfiguration```. 

this is the code I use:

```java
fotoapparat.getCapabilities().whenDone(capabilities -> {
                    Set<Resolution> previewResolutions = capabilities.getPreviewResolutions();
                    Set<Resolution> pictureResolutions = capabilities.getPictureResolutions();

                    fotoapparat.updateConfiguration(UpdateConfiguration.builder()
                            .photoResolution(new ResolutionSelectorHelper().highestResolution(prepareListForRatio(pictureResolutions)))
                            .previewResolution(new ResolutionSelectorHelper().highestResolution(prepareListForRatio(previewResolutions)))
                            .focusMode(SelectorsKt.firstAvailable(
                                    FocusModeSelectorsKt.continuousFocusPicture(),
                                    FocusModeSelectorsKt.autoFocus(),
                                    FocusModeSelectorsKt.fixed()
                            ))
                            .jpegQuality(JpegQualitySelectorsKt.manualJpegQuality(100))
                            .flash(off())
                            .build());
                });
```

this is the method ```highestResolution``` in ```ResolutionSelectorHelper```

```java
fun highestResolution(list: Collection<Resolution>): ResolutionSelector = {
        Collections.max(list) { left, right -> Integer.compare(left.width, right.width) }
    }
```
and method ```prepareListForRatio```

```java
private Collection<Resolution> prepareListForRatio(Collection<Resolution> sizes) {
        double ratio = 4.0 / 3.0;
        Iterator<Resolution> iterator = sizes.iterator();

        while (iterator.hasNext()) {
            Resolution resolution = iterator.next();

            if (Math.abs((double) resolution.width / (double) resolution.height - ratio) > 0.02) {
                iterator.remove();
            }
        }

        return sizes;
    }
```

The problem is that the photo resolution is changed, but preview stays the same.

wellbranding, 2019-04-12:
Facing same issue. Are there any workarounds?

PawelTypiak, 2019-04-12:
Yep, i found one.

You have to update also the cameraView like:

`cameraView().setPreviewResolution(new Resolution(previewResolution.height,previewResolution.width));`

just remember to switch the width and height of your preview `Resolution`

wellbranding, 2019-04-12:
Thanks! I also saw that method, but it did not worked, because I did not switched values. What I did was the following: stop fotoapparat instance, remove fotoapparat instance, **create a new one** with desired resolution and then call ### fotoapparat.start() o new one. Your solution seems much better. 
Does it work well on all devices, also? 

I don't get why you need to switch Resolution values?

PawelTypiak, 2019-04-12:
I'm glad that I helped.

My app has over 100k users and I didn't have any issues with this solution whatsoever. I guess cameraView has implemented `Resolution` in a reversed way.