RedApparat/Fotoapparat #341: quick capture mode- Sample Code

sharma4frnds, 2019-01-28:
Hi,
   It would be great if you can share sample code for quick capture mode. 

I am using fotoapparat.takePicture(); but it is taking avg: 1.85secs in capturing images while switching camera and capturing images. Or is there any other way where we can reduce time ?

wellbranding, 2019-02-02:
Yes also interested. I have created custom frame processor but I think it is bad...

sharma4frnds, 2019-02-04:
So please suggest, how we can use quick capture feature?

huckw, 2019-07-22:
I have the same question.  I have a frame processor which allows me to grab preview frames quickly BUT these are not the full resolution images that I get from takePicture().  I Need to snag about 3 images and don't need to write them to file (I'll figure out which of the 3 is best and save this one to file later).  Any tips you have are appreciated!

nilabh, 2019-07-23:
> Hi,
> It would be great if you can share sample code for quick capture mode.
> 
> I am using fotoapparat.takePicture(); but it is taking avg: 1.85secs in capturing images while switching camera and capturing images. Is there any other way where we can reduce time ?

@sharma4frnds u can just keep clicking picture and create listeners to write image to file in a separate thread.. Even if u try to do quick capture in default cameras, the preview in the bottom right/left loads in a separate thread and takes time to load.
Also it is taking time because you must be trying the highest resolution . If you can compromise somewhat on the resolution then you can do it quite fast. I've a resolution max of width 1280 and i get the image in around 700ms

nilabh, 2019-07-23:
> I have the same question. I have a frame processor which allows me to grab preview frames quickly BUT these are not the full resolution images that I get from takePicture(). I Need to snag about 3 [...]

@huckw if you are trying to capture high resolution images , then not saving to file and keeping in memory is not the best solution. I would suggest you  write to file and delete after selection, that won't be an overhead and will improve performance.

sharma4frnds, 2019-07-24:
What is the capture time if i capture images from both the camera using
single ISP devices.

Ex.
Camera 1 -> Open -> capture image -> release
Camera 2 -> Open -> capture image -> release

Combined time for this process ?