RedApparat/Fotoapparat #344: How would I scale to get a fixed image resolution (width=768) on any smartphone?

murathada, 2019-02-05:
I take pictures with the fotoapparat.io and in the `whenDone()` I have my `BitmapPhoto` object. 

Since I dont need that those big resolutions I would be satisfied with photos where the width has 768 pixels. The height would depend on smartphone cameras aspect ratio. On 4:3 cameras the height would be 576 pixels. 

So far so good.

To scale the image down I use
```
 photoResult.toBitmap(
                    new Function1<Resolution, Resolution>() {
                        @Override
                        public Resolution invoke(Resolution resolution) {
                            float scale = (float) 768 / resolution.width;
                            return scaled(scale).invoke(resolution);
                        }
                    }
            )
```
But on some devices I get lower or bigger resolutions as an end photo. 

**How can I get fixed width x (e.g. 768) pixels, for any smaertphones, with any camera resolutions and with any screen sizes?**

murathada, 2019-02-05:
Also 
```
            photoResult.toBitmap(
                new Function1<Resolution, Resolution>() {
                    @Override
                    public Resolution invoke(Resolution resolution) {
                        int height = 1024;
                        int width = 768;
                        Resolution r = new Resolution(width, height);
                        return scaled(1.0F).invoke(r);
                    }
                }
            )
```

did not work very well.

murathada, 2019-02-08:
So no solution here?

Diolor, 2019-03-02:
You cannot. You can always take higher resolution images and scale them down but this library has nothing to do with image scaling

closed 2019-03-02 by Diolor