RedApparat/Fotoapparat #345: How can i apply converted frame to preview/photoResult

mym0404, 2019-02-09:
#### What are you trying to achieve or the steps to reproduce?

Hi I want to apply filter/flip/other image processing to preview/photoResult

I know this things are not work of this library.

I want to know is that 

1. I converted byte array to converted byte array or bitmap in frameProcessing{}

2. how can i apply this image to preview and captured photo?

I know this is stupid question... Sorry about that

#### How did you initialize FA?

#### What was the result you received?

convert frame.image to converted byte array not affected..

#### What did you expect?

#### Context:

* *FA version*:
* *Devices/APIs affected*:
* *any other relevant information*:

mym0404, 2019-02-19:
I solved my problem with Frame Processor.

Frame Processor can't handle preview itself.

So I did like below.

1. convert byte[] (YUV format) to image 

2. processing image (OpenGL)

3. Show Image using other GLSurfaceView

4. Turn off CameraView of this library (turn off setPreviewTexture & setPreviewSurface?)

Then, I can receive only byte[] data from this great library.

kleysonr, 2019-04-10:
Hi @mym0404,

May you share the final layout and your FrameProcessor class ? I am a newbie on android dev and I need to achieve the same as you.

Thanks.

closed 2024-03-19 by mym0404