RedApparat/Fotoapparat #353: Obtain horizontal field of view

nickw1, 2019-03-01:
Really nice API, makes it very easy to take photos. Was easy to figure out how to stream the camera feed to a GLSurfaceView.Renderer too.

One thing which appears to be missing at the moment is an API call to obtain the field of view (or the focal length and sensor size, which can be used to calculate it). I'm developing an augmented reality app at the moment and would be really nice to get the FOV so this can be used to properly align the AR data with the camera view. My work-round is to use Camera2's CameraCharacteristics just for this feature, which seems to work (on my device) but necessitates minSdk of 21 so would be nice (if no technical obstacles) to see this in FA.

Many thanks for your work though! :-)

Diolor, 2019-03-02:
Hey, I see [getFocalLength](https://developer.android.com/reference/android/hardware/Camera.Parameters.html#getFocalLength()) from Camera 1. Not sure if that gives you enough info to work with. If you like to implement it in FA, feel free to PR us! :)

nickw1, 2019-03-03:
Hello Dionysis,

OK - thanks.
Will try to if I have the time!