RedApparat/Fotoapparat #355: Can't make a photo on an OnePlus 6T with Android 9

wolf81, 2019-03-04:
Please check if your issue exists.

Issues should only be posted after you have been able to reproduce them and confirm that they are either a missing functionality or a bug. 

#### What are you trying to achieve or the steps to reproduce?

Trying to make a photo on a OnePlus 6T device with Android 9. I am testing this device using AppCenter, as I don't own one myself. An exception occurs trying to make a photo regarding the orientation.

```java
            PhotoResult photoResult = camera.takePicture();

            // PLEASE NOTE: We perform the save on a separate thread manually instead of using the
            // saveToFile() method, since the saveToFile() method doesn't properly log exceptions
            Thread thread = new Thread() {
                @Override
                public void run() {
                    try {
                        photoResult.saveToFile(imageFile).await();
                    } catch (Exception e) {
                        Log.e(TamperSecure.TAG, "*** ERROR: " + e.toString() );
                        e.printStackTrace();
                }
            };
            thread.start();
```

#### How did you initialize FA?

```java
        this.camera = Fotoapparat
                .with(this)
                .into(cameraView)
                .lensPosition(SelectorsKt.firstAvailable(
                        LensPositionSelectorsKt.back(),
                        LensPositionSelectorsKt.front(),
                        LensPositionSelectorsKt.external()
                ))
                .previewResolution((resolutions) -> getResolutionWithMp(1.f, resolutions))
                .photoResolution((resolutions) -> getResolutionWithMp(5.f, resolutions))
                .previewScaleType(ScaleType.CenterCrop)
                .flash(off())
                .focusMode(SelectorsKt.firstAvailable(
                        FocusModeSelectorsKt.continuousFocusPicture(),
                        FocusModeSelectorsKt.autoFocus(),
                        FocusModeSelectorsKt.fixed(),
                        FocusModeSelectorsKt.infinity()
                ))
                .logger(LoggersKt.logcat())
                .cameraErrorCallback((exception) -> {
                    Log.e(TamperSecure.TAG, exception.toString());
                })
                .build();
```

#### What was the result you received?

```
D/Fotoapparat( 7056): Fotoapparat: takePictureD/Fotoapparat( 7056): CameraDevice: takePhotoE/libc    ( 7056): Access denied finding property "vendor.debug.egl.swapinterval"D/Fotoapparat( 7056): Couldn't deliver pending result: Operation failed internally.W/RenderThread( 7056): type=1400 audit(0.0:1019): avc: denied { read } for name="u:object_r:vendor_default_prop:s0" dev="tmpfs" ino=16873 scontext=u:r:untrusted_app:s0:c166,c262,c512,c768 tcontext=u:object_r:vendor_default_prop:s0 tclass=file permissive=0E/TamperSecure( 7056): *** ERROR: java.util.concurrent.ExecutionException: java.util.concurrent.ExecutionException: kotlin.UninitializedPropertyAccessException: lateinit property imageOrientation has not been initializedW/System.err( 7056): java.util.concurrent.ExecutionException: java.util.concurrent.ExecutionException: kotlin.UninitializedPropertyAccessException: lateinit property imageOrientation has not been initializedW/System.err( 7056): 	at java.util.concurrent.FutureTask.report(FutureTask.java:123)W/System.err( 7056): 	at java.util.concurrent.FutureTask.get(FutureTask.java:193)W/System.err( 7056): 	at io.fotoapparat.result.PendingResult.await(PendingResult.kt:55)W/System.err( 7056): 	at com.peir.photector.CameraActivity$1.run(CameraActivity.java:217)W/System.err( 7056): Caused by: java.util.concurrent.ExecutionException: kotlin.UninitializedPropertyAccessException: lateinit property imageOrientation has not been initializedW/System.err( 7056): 	at java.util.concurrent.FutureTask.report(FutureTask.java:123)W/System.err( 7056): 	at java.util.concurrent.FutureTask.get(FutureTask.java:193)W/System.err( 7056): 	at io.fotoapparat.result.PendingResult$transform$transformTask$1.call(PendingResult.kt:36)W/System.err( 7056): 	at java.util.concurrent.FutureTask.run(FutureTask.java:266)W/System.err( 7056): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)W/System.err( 7056): 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)W/System.err( 7056): 	at java.lang.Thread.run(Thread.java:764)W/System.err( 7056): Caused by: kotlin.UninitializedPropertyAccessException: lateinit property imageOrientation has not been initializedW/System.err( 7056): 	at io.fotoapparat.hardware.CameraDevice.takePhoto(CameraDevice.kt:139)W/System.err( 7056): 	at io.fotoapparat.routine.photo.TakePhotoRoutineKt$takePhoto$1.invokeSuspend(TakePhotoRoutine.kt:15)W/System.err( 7056): 	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:32)W/System.err( 7056): 	at kotlinx.coroutines.DispatchedTask$DefaultImpls.run(Dispatched.kt:235)W/System.err( 7056): 	at kotlinx.coroutines.DispatchedContinuation.run(Dispatched.kt:81)W/System.err( 7056): 	at kotlinx.coroutines.EventLoopBase.processNextEvent(EventLoop.kt:123)W/System.err( 7056): 	at kotlinx.coroutines.BlockingCoroutine.joinBlocking(Builders.kt:69)I/TamperSecure( 7056): *** STOP CAMERAW/System.err( 7056): 	at kotlinx.coroutines.BuildersKt__BuildersKt.runBlocking(Builders.kt:45)W/System.err( 7056): 	at kotlinx.coroutines.BuildersKt.runBlocking(Unknown Source:1)W/System.err( 7056): 	at kotlinx.coroutines.BuildersKt__BuildersKt.runBlocking$default(Builders.kt:35)W/System.err( 7056): 	at kotlinx.coroutines.BuildersKt.runBlocking$default(Unknown Source:1)
```

#### What did you expect?

I would have expected to be able to save the file successfully.

#### Context:

* *FA version*: Android 9
* *Devices/APIs affected*: OnePlus 6T
* *any other relevant information*:

Diolor, 2019-03-04:
Hey, thanks for reporting. Are you by any chance calling take picture before starting FA?

wolf81, 2019-03-04:
I believe Fotoapparat should have been started already. I'll show a more complete version of the log. Please note init is called previously.

```
D/Fotoapparat( 7056): Device: updateLensPositionSelector
D/Fotoapparat( 7056): Fotoapparat: <init>
D/LifecycleMonitor( 7056): Lifecycle status change: com.peir.photector.CameraActivity@d9429a5 in: CREATED
D/LifecycleMonitor( 7056): running callback: androidx.test.rule.ActivityTestRule$LifecycleCallback@becf7e9
D/LifecycleMonitor( 7056): callback completes: androidx.test.rule.ActivityTestRule$LifecycleCallback@becf7e9
D/Fotoapparat( 7056): Fotoapparat: start
D/LifecycleMonitor( 7056): Lifecycle status change: com.peir.photector.CameraActivity@d9429a5 in: STARTED
D/LifecycleMonitor( 7056): running callback: androidx.test.rule.ActivityTestRule$LifecycleCallback@becf7e9
D/LifecycleMonitor( 7056): callback completes: androidx.test.rule.ActivityTestRule$LifecycleCallback@becf7e9
D/Fotoapparat( 7056): Fotoapparat: getCapabilities
D/LifecycleMonitor( 7056): Lifecycle status change: com.peir.photector.CameraActivity@d9429a5 in: RESUMED
D/LifecycleMonitor( 7056): running callback: androidx.test.rule.ActivityTestRule$LifecycleCallback@becf7e9
D/LifecycleMonitor( 7056): callback completes: androidx.test.rule.ActivityTestRule$LifecycleCallback@becf7e9
D/Fotoapparat( 7056): Device: selectCamera
D/Fotoapparat( 7056): CameraDevice: open
W/pool-7-thread-1( 7056): type=1400 audit(0.0:1008): avc: denied { read } for name="u:object_r:vendor_default_prop:s0" dev="tmpfs" ino=16873 scontext=u:r:untrusted_app:s0:c166,c262,c512,c768 tcontext=u:object_r:vendor_default_prop:s0 tclass=file permissive=0
E/libc    ( 7056): Access denied finding property "vendor.camera.hal1.packagelist"
I/CameraService(  930): CameraService::connect call (PID -1 "com.peir.tampersecure_example", camera ID 0) for HAL version default and Camera API version 1
I/CameraService(  930): CameraService::connect Bypass active check: com.peir.tampersecure_example (active: 1)
D/MediaPlayerService(  944): LF player type = 4
D/NuPlayerDriver(  944): NuPlayerDriver(0xf0793480) created, clientPid(930)
E/FileSource(  944): Failed to open file '/product/media/audio/ui/camera_click.ogg'. (No such file or directory)
E/GenericSource(  944): Failed to create data source!
D/NuPlayerDriver(  944): notifyListener_l(0xf0793480), (100, 1, -2147483648, -1), loop setting(0, 0)
E/MediaPlayerNative(  930): error (1, -2147483648)
E/CameraService(  930): Failed to load CameraService sounds: /product/media/audio/ui/camera_click.ogg
D/NuPlayerDriver(  944): reset(0xf0793480) at state 2
D/NuPlayerDriver(  944): notifyListener_l(0xf0793480), (8, 0, 0, -1), loop setting(0, 0)
W/AMessage(  944): failed to post message as target looper for handler 0 is gone.
D/NuPlayerDriver(  944): notifyResetComplete(0xf0793480)
D/MediaPlayerService(  944): LF player type = 4
D/NuPlayerDriver(  944): NuPlayerDriver(0xf0793500) created, clientPid(930)
I/QComExtractorFactory(  936):  Sniff OGG success <==
I/MMParserExtractor(  936): Created(0x7d0a4c7700)
D/MMParserExtractor(  936): MMParserExtractor create m_eFileFormat 16
W/BpMediaSource(  944): ignoring potentially modified MetaData from start
W/BpMediaSource(  944): input:
I/MetaDataBase(  944): bytm: (int32_t) 2
W/BpMediaSource(  944): output:
I/MetaDataBase(  944): bytm: (int32_t) 0
D/NuPlayerDriver(  944): notifyListener_l(0xf0793500), (1, 0, 0, -1), loop setting(0, 0)
D/MediaPlayerService(  944): LF player type = 4
D/NuPlayerDriver(  944): NuPlayerDriver(0xf0793580) created, clientPid(930)
E/FileSource(  944): Failed to open file '/product/media/audio/ui/VideoRecord.ogg'. (No such file or directory)
E/GenericSource(  944): Failed to create data source!
D/NuPlayerDriver(  944): notifyListener_l(0xf0793580), (100, 1, -2147483648, -1), loop setting(0, 0)
E/MediaPlayerNative(  930): error (1, -2147483648)
E/CameraService(  930): Failed to load CameraService sounds: /product/media/audio/ui/VideoRecord.ogg
D/NuPlayerDriver(  944): reset(0xf0793580) at state 2
D/NuPlayerDriver(  944): notifyListener_l(0xf0793580), (8, 0, 0, -1), loop setting(0, 0)
W/AMessage(  944): failed to post message as target looper for handler 0 is gone.
D/NuPlayerDriver(  944): notifyResetComplete(0xf0793580)
D/MediaPlayerService(  944): LF player type = 4
D/NuPlayerDriver(  944): NuPlayerDriver(0xf0793600) created, clientPid(930)
I/QComExtractorFactory(  936):  Sniff OGG success <==
I/MMParserExtractor(  936): Created(0x7d0a4c77e0)
D/MMParserExtractor(  936): MMParserExtractor create m_eFileFormat 16
W/BpMediaSource(  944): ignoring potentially modified MetaData from start
W/BpMediaSource(  944): input:
I/MetaDataBase(  944): bytm: (int32_t) 2
W/BpMediaSource(  944): output:
I/MetaDataBase(  944): bytm: (int32_t) 0
D/NuPlayerDriver(  944): notifyListener_l(0xf0793600), (1, 0, 0, -1), loop setting(0, 0)
D/MediaPlayerService(  944): LF player type = 4
D/NuPlayerDriver(  944): NuPlayerDriver(0xf0793b80) created, clientPid(930)
E/FileSource(  944): Failed to open file '/product/media/audio/ui/VideoStop.ogg'. (No such file or directory)
E/GenericSource(  944): Failed to create data source!
[... 189 lines cut ...]
I/ViewInteraction( 7056): Performing 'single click' action on view with id: com.peir.tampersecure_example:id/camera_view
E/libc    ( 7056): Access denied finding property "vendor.debug.egl.swapinterval"
[... cut ...]
W/System.err( 7056): 	at java.util.concurrent.FutureTask.get(FutureTask.java:193)
W/System.err( 7056): 	at io.fotoapparat.result.PendingResult$transform$transformTask$1.call(PendingResult.kt:36)
W/System.err( 7056): 	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
W/System.err( 7056): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
W/System.err( 7056): 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
W/System.err( 7056): 	at java.lang.Thread.run(Thread.java:
```