RedApparat/Fotoapparat #357: Flood of error messages as soon as the camera opens

Tsabary, 2019-03-12:
As soon as I open the camera fragment that hosts the preview, my logcat starts writing a flow of repeating error messages that don't stop. This is the messages

`2019-03-12 18:00:35.760 732-25845/? E/mm-camera: <SENSOR><ERROR> 497: port_sensor_handle_aec_update: flash_process  est_state= 0 ,state =0 ,torch_on =0, partial_flash_frame_skip =1, is_dual_cam = 0
2019-03-12 18:00:35.760 732-25846/? E/libc: Access denied finding property "persist.camera.pdaf_logging"
2019-03-12 18:00:35.760 732-25846/? E/libc: Access denied finding property "persist.camera.pdaf_logging"
2019-03-12 18:00:35.760 732-25846/? E/libc: Access denied finding property "persist.camera.pdaf_profiling"
2019-03-12 18:00:35.791 732-25846/? E/libc: Access denied finding property "persist.camera.pdaf_profiling"
2019-03-12 18:00:35.791 732-25846/? E/libc: Access denied finding property "persist.camera.pdaf_logging"
2019-03-12 18:00:35.791 732-25846/? E/libc: Access denied finding property "persist.camera.pdaf_profiling"
2019-03-12 18:00:35.791 732-25846/? E/libc: Access denied finding property "persist.camera.pdaf_profiling"
2019-03-12 18:00:35.791 732-25846/? E/libc: Access denied finding property "persist.camera.pdaf_logging"
2019-03-12 18:00:35.792 732-25828/? E/mm-camera: <SENSOR><ERROR> 497: port_sensor_handle_aec_update: flash_process  est_state= 0 ,state =0 ,torch_on =0, partial_flash_frame_skip =1, is_dual_cam = 0
2019-03-12 18:00:35.794 732-25846/? E/libc: Access denied finding property "persist.camera.pdaf_logging"
2019-03-12 18:00:35.794 732-25846/? E/libc: Access denied finding property "persist.camera.pdaf_logging"
2019-03-12 18:00:35.794 732-25846/? E/libc: Access denied finding property "persist.camera.pdaf_profiling"`

wellbranding, 2019-03-12:
Does camera work properly?

Tsabary, 2019-03-13:
It does but the unstoppable stream of errors is extremely abnormal. I'm talking so fast the if it wouldn't have been the same message I would not have been able to read even a word as it goes.

Diolor, 2019-03-13:
This doesn't look connected to FA. Simply a device lower level log

closed 2019-03-13 by Diolor

smitapaul147, 2019-08-07:
i faced same problem.what can i do?