RedApparat/Fotoapparat #364: error: cannot find symbol class OnProgressChanged

kleysonr, 2019-04-10:
#### What are you trying to achieve or the steps to reproduce?

I'm trying to use the Fotoapparat in a java project. I copied the res folders from the git sample as well as ActivityJava.java and PermissionsDelegate.java files.

After fixing all the imports, I'm getting "Cannot resolve symbol OnProgressChanged" in the following method.

```java
    private void zoomSeekBar() {
        SeekBar seekBar = findViewById(R.id.zoomSeekBar);

        seekBar.setOnSeekBarChangeListener(new OnProgressChanged() {
            @Override
            public void onProgressChanged(SeekBar seekBar, int progress, boolean fromUser) {
                fotoapparat.setZoom(progress / (float) seekBar.getMax());
            }
        });
    }
```

#### Context:

* *FA version*: 2.7.0
* *Devices/APIs affected*: API-21

Diolor, 2019-04-10:
Well apparently you don't have a xml with zoomSeekBar id. This is not a library issue

closed 2019-04-10 by Diolor

kleysonr, 2019-04-10:
Hi @Diolor ,

My project is basically a clone of the Fotoapparat's sample (https://github.com/RedApparat/Fotoapparat/tree/master/sample) but using the java version.

And there is the zoomSeekBar id in the activity_main.xml file (.xml below).

Might it be an issue in java sample instead ?

```
<?xml version="1.0" encoding="utf-8"?>
<android.support.v7.widget.FitWindowsFrameLayout xmlns:android="http://schemas.android.com/apk/res/android"
    xmlns:tools="http://schemas.android.com/tools"
    android:layout_width="match_parent"
    android:layout_height="match_parent"
    android:background="#282828"
    tools:activity="io.fotoapparat.sample.MainActivity">

    <io.fotoapparat.view.CameraView
        android:id="@+id/cameraView"
        android:layout_width="match_parent"
        android:layout_height="match_parent"
        android:visibility="gone"
        tools:layout_editor_absoluteX="8dp"
        tools:layout_editor_absoluteY="8dp"
        tools:visibility="visible">

        <!--Optionally add tap to focus-->
        <io.fotoapparat.view.FocusView
            android:id="@+id/focusView"
            android:layout_width="match_parent"
            android:layout_height="match_parent" />

    </io.fotoapparat.view.CameraView>

    <TextView
        android:id="@+id/no_permission"
        android:layout_width="wrap_content"
        android:layout_height="wrap_content"
        android:layout_gravity="center_horizontal|bottom"
        android:layout_marginBottom="100dp"
        android:text="No Camera permission granted."
        android:textSize="20sp"
        android:visibility="gone"
        tools:ignore="HardcodedText"
        tools:visibility="visible" />

    <FrameLayout
        android:layout_width="match_parent"
        android:layout_height="match_parent"
        android:fitsSystemWindows="true">

        <FrameLayout
            android:layout_width="match_parent"
            android:layout_height="wrap_content"
            android:layout_gravity="top">

            <android.support.v7.widget.SwitchCompat
                android:id="@+id/torchSwitch"
                android:layout_width="wrap_content"
                android:layout_height="wrap_content"
                android:layout_gravity="left|center_vertical"
                android:padding="20dp"
                tools:ignore="RtlHardcoded" />

            <SeekBar
                android:id="@+id/zoomSeekBar"
                android:layout_width="200dp"
                android:layout_height="wrap_content"
                android:layout_gravity="center" />

            <ImageView
                android:id="@+id/switchCamera"
                android:layout_width="wrap_content"
                android:layout_height="wrap_content"
                android:layout_gravity="right|center_vertical"
                android:padding="20dp"
                android:src="@drawable/ic_autorenew_white"
                tools:ignore="RtlHardcoded" />

        </FrameLayout>

        <TextView
            android:id="@+id/zoomLvl"
            android:layout_width="wrap_content"
            android:layout_height="wrap_content"
            android:layout_gravity="center_horizontal|top"
            android:layout_marginTop="50dp"
            android:textColor="#FFF"
            android:textSize="20sp"
            tools:text="2.4" />

        <ImageView
            android:id="@+id/result"
            android:layout_width="150dp"
            android:layout_height="150dp"
            android:layout_gravity="bottom|left"
            tools:ignore="ContentDescription,RtlHardcoded" />

        <ImageView
            android:id="@+id/capture"
            android:layout_width="70dp"
            android:layout_height="70dp"
            android:layout_gravity="bottom|center_horizontal"
            android:layout_marginBottom="10dp"
            android:src="@drawable/capture"
            tools:ignore="ContentDescription" />
    </FrameLayout>

</android.support.v7.widget.FitWindowsFrameLayout>

```

isabsent, 2019-12-31:
Just change `zoomSeekBar` method definition in `ActivityJava`:

    private void zoomSeekBar() {
        SeekBar seekBar = findViewById(R.id.zoomSeekBar);

        seekBar.setOnSeekBarChangeListener(new SeekBar.OnSeekBarChangeListener() {

            @Override
            public void onProgressChanged(SeekBar seekBar, int progress, boolean fromUser) {
                fotoapparat.setZoom(progress / (float) seekBar.getMax());
            }

            @Override
            public void onStartTrackingTouch(SeekBar seekBar) {

            }

            @Override
            public void onStopTrackingTouch(SeekBar seekBar) {

            }
        });
    }