RedApparat/Fotoapparat #367: Enable negative effect mode

funkyidol, 2019-04-22:
I want to apply the Negative effect mode similar to whats available in Camera2 API. How can I acheive the same effect using Fotoapparat?

```java
previewRequestBuilder.set(CaptureRequest.CONTROL_EFFECT_MODE,
                        CaptureRequest.CONTROL_EFFECT_MODE_NEGATIVE)
```
* *FA version*: 2.7.0

funkyidol, 2019-04-22:
I see that you have a [test in your code](https://github.com/RedApparat/Fotoapparat/blob/master/fotoapparat/src/test/java/io/fotoapparat/parameter/extract/RawValuesExtractorTest.kt) which actually tries to fetch the effects and verify if they are available. But I dont see them being extracted in the main library or an option to set the said parameter type. Can something similar be used to apply the negative filter directly as a `Camera.Parameter` so that both the preview and capture frames will have effect enabled without additional processing?

If yes, I would like to implement this (if its not already there) on a fork and send you a PR for the same. Though some additional guidance regarding the same will not hurt.

coudys, 2023-08-03:
I found one app on google play store that can do the negative effect on my phne but I dont have source code. For now I just chnage the RGB values using getpixels() and for loop

if (negativeTrue)
	{
	// Apply the negative effect to the camera feed texture
	Color[] colors = backCameraTexture.GetPixels();
	for (int i = 0; i < colors.Length; i++)
		{
			colors[i].r = 1.0f - colors[i].r;
			colors[i].g = 1.0f - colors[i].g;
			colors[i].b = 1.0f - colors[i].b;
		}
		negativeTexture.SetPixels(colors);
		negativeTexture.Apply();
		targetImage.texture = negativeTexture;
	}
	else {
		targetImage.texture = backCameraTexture;
	}