RedApparat/Fotoapparat #370: Compatibility with Gradle 5 regarding dependencies

avianey, 2019-05-02:
Latest Stable Android Studio release (3.4 so far) and tooling (com.android.tools.build:gradle:3.4.0) uses gradle 5.1.1 which breaks maven dependency management : 

- https://docs.gradle.org/current/userguide/upgrading_version_4.html#rel5.0:pom_compile_runtime_separation

When using the 2.7.0 version of fotoappart using above configuration like this (in JAVA)

```java
Fotoapparat
    .with(getContext())
    .into((CameraView) view)
    .previewScaleType(ScaleType.CenterCrop)
    .photoResolution(ResolutionSelectorsKt.highestResolution())
    .lensPosition(LensPositionSelectorsKt.back())
    .focusMode(SelectorsKt.firstAvailable(
            FocusModeSelectorsKt.continuousFocusPicture(),
            FocusModeSelectorsKt.autoFocus(),
            FocusModeSelectorsKt.fixed()
    ))
    .flash(SelectorsKt.firstAvailable(
            FlashSelectorsKt.autoRedEye(),
            FlashSelectorsKt.autoFlash(),
            FlashSelectorsKt.torch()
    ))
    .frameProcessor(frameProcessor)
    .logger(LoggersKt.logcat())
```

Compilation fails :

```
error: cannot access Function1
class file for kotlin.jvm.functions.Function1 not found
```

Because **runtime** dependencies are no more imported by gradle 5...
...and you fotoappart-2.7.0.pom file uses **runtime** scope.

```xml
    <dependency>
      <groupId>androidx.annotation</groupId>
      <artifactId>annotation</artifactId>
      <version>1.0.0</version>
      <scope>runtime</scope>
    </dependency>
    <dependency>
      <groupId>androidx.exifinterface</groupId>
      <artifactId>exifinterface</artifactId>
      <version>1.0.0</version>
      <scope>runtime</scope>
    </dependency>
    <dependency>
      <groupId>org.jetbrains.kotlin</groupId>
      <artifactId>kotlin-stdlib</artifactId>
      <version>1.3.0</version>
      <scope>runtime</scope>
    </dependency>
    <dependency>
      <groupId>org.jetbrains.kotlinx</groupId>
      <artifactId>kotlinx-coroutines-core</artifactId>
      <version>1.0.0</version>
      <scope>runtime</scope>
    </dependency>
```

#### What did you expect?

Dependencies are imported when project uses :

```
implementation 'io.fotoapparat:fotoapparat:2.7.0'
```

For this, using gradle 5, the pom needs to use **compile** scope now instead of **runtime**.
And needs to be re generated (using an up to date library that understand **implementation** scope)

#### Context:

* 2.7.0
* Android Studio 3.4 / Gradle 5 / Java project

avianey, 2019-05-02:
Quick / Dirty fix is to manually include dependencies :
```
implementation 'io.fotoapparat:fotoapparat:2.7.0'
// add missing dependencies
implementation "androidx.annotation:annotation:1.0.2"
implementation "androidx.exifinterface:exifinterface:1.0.0"
implementation "org.jetbrains.kotlin:kotlin-stdlib:1.3.30"
```

vhontar, 2019-05-08:
Hi, I have another problem after adding this quick /dirty fix above:
:app:checkDevelopDebugDuplicateClasses

`org.gradle.api.tasks.TaskExecutionException: Execution failed for task ':app:checkDevelopDebugDuplicateClasses'.
	at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.execute(ExecuteActionsTaskExecuter.java:95)
	at org.gradle.api.internal.tasks.execution.ResolveTaskOutputCachingStateExecuter.execute(ResolveTaskOutputCachingStateExecuter.java:91)
	at org.gradle.api.internal.tasks.execution.ValidatingTaskExecuter.execute(ValidatingTaskExecuter.java:57)
	at org.gradle.api.internal.tasks.execution.SkipEmptySourceFilesTaskExecuter.execute(SkipEmptySourceFilesTaskExecuter.java:119)
	at org.gradle.api.internal.tasks.execution.ResolvePreviousStateExecuter.execute(ResolvePreviousStateExecuter.java:43)
	at org.gradle.api.internal.tasks.execution.CleanupStaleOutputsExecuter.execute(CleanupStaleOutputsExecuter.java:93)
	at org.gradle.api.internal.tasks.execution.FinalizePropertiesTaskExecuter.execute(FinalizePropertiesTaskExecuter.java:45)
	at org.gradle.api.internal.tasks.execution.ResolveTaskArtifactStateTaskExecuter.execute(ResolveTaskArtifactStateTaskExecuter.java:94)
	at org.gradle.api.internal.tasks.execution.SkipTaskWithNoActionsExecuter.execute(SkipTaskWithNoActionsExecuter.java:56)
	at org.gradle.api.internal.tasks.execution.SkipOnlyIfTaskExecuter.execute(SkipOnlyIfTaskExecuter.java:55)
	at org.gradle.api.internal.tasks.execution.CatchExceptionTaskExecuter.execute(CatchExceptionTaskExecuter.java:36)
	at org.gradle.api.internal.tasks.execution.EventFiringTaskExecuter$1.executeTask(EventFiringTaskExecuter.java:67)
	at org.gradle.api.internal.tasks.execution.EventFiringTaskExecuter$1.call(EventFiringTaskExecuter.java:52)
	at org.gradle.api.internal.tasks.execution.EventFiringTaskExecuter$1.call(EventFiringTaskExecuter.java:49)
	at org.gradle.internal.operations.DefaultBuildOperationExecutor$CallableBuildOperationWorker.execute(DefaultBuildOperationExecutor.java:315)
	at org.gradle.internal.operations.DefaultBuildOperationExecutor$CallableBuildOperationWorker.execute(DefaultBuildOperationExecutor.java:305)
	at org.gradle.internal.operations.DefaultBuildOperationExecutor.execute(DefaultBuildOperationExecutor.java:175)
	at org.gradle.internal.operations.DefaultBuildOperationExecutor.call(DefaultBuildOperationExecutor.java:101)
	at org.gradle.internal.operations.DelegatingBuildOperationExecutor.call(DelegatingBuildOperationExecutor.java:36)
	at org.gradle.api.internal.tasks.execution.EventFiringTaskExecuter.execute(EventFiringTaskExecuter.java:49)
	at org.gradle.execution.plan.LocalTaskNodeExecutor.execute(LocalTaskNodeExecutor.java:43)
	at org.gradle.execution.taskgraph.DefaultTaskExecutionGraph$InvokeNodeExecutorsAction.execute(DefaultTaskExecutionGraph.java:355)
	at org.gradle.execution.taskgraph.DefaultTaskExecutionGraph$InvokeNodeExecutorsAction.execute(DefaultTaskExecutionGraph.java:343)
	at org.gradle.execution.taskgraph.DefaultTaskExecutionGraph$BuildOperationAwareExecutionAction.execute(DefaultTaskExecutionGraph.java:336)
	at org.gradle.execution.taskgraph.DefaultTaskExecutionGraph$BuildOperationAwareExecutionAction.execute(DefaultTaskExecutionGraph.java:322)
	at org.gradle.execution.plan.DefaultPlanExecutor$ExecutorWorker$1.execute(DefaultPlanExecutor.java:134)
	at org.gradle.execution.plan.DefaultPlanExecutor$ExecutorWorker$1.execute(DefaultPlanExecutor.java:129)
	at org.gradle.execution.plan.DefaultPlanExecutor$ExecutorWorker.execute(DefaultPlanExecutor.java:202)
	at org.gradle.execution.plan.DefaultPlanExecutor$ExecutorWorker.executeNextNode(DefaultPlanExecutor.java:193)
	at org.gradle.execution.plan.DefaultPlanExecutor$ExecutorWorker.run(DefaultPlanExecutor.java:129)
	at org.gradle.execution.plan.DefaultPlanExecutor.process(DefaultPlanExecutor.java:74)
	at org.gradle.execution.taskgraph.DefaultTaskExecutionGraph.executeWithServices(DefaultTaskExecutionGraph.java:178)
	at org.gradle.execution.taskgraph.DefaultTaskExecutionGraph.execute(DefaultTaskExecutionGraph.java:154)
	at org.gradle.execution.SelectedTaskExecutionAction.execute(SelectedTaskExecutionAction.java:41)
	at org.gradle.execution.DefaultBuildExecuter.execute(DefaultBuildExecuter.java:40)
	at org.gradle.execution.DefaultBuildExecuter.access$000(DefaultBuildExecuter.java:24)
	at org.gradle.execution.DefaultBuildExecuter$1.proceed(DefaultBuildExecuter.java:46)
	at org.gradle.execution.DryRunBuildExecutionAction.execute(DryRunBuildExecutionAction.java:49)
	at org.gradle.execution.DefaultBuildExecuter.execute(DefaultBuildExecuter.java:40)
	at org.gradle.execution.DefaultBuildExecuter.execute(DefaultBuildExecuter.java:33)
	at org.gradle.initialization.DefaultGradleLauncher$ExecuteTasks.run(DefaultGradleLauncher.java:383)
	at org.gradle.internal.operations.DefaultBuildOperationExecutor$RunnableBuildOperationWorker.execute(DefaultBuildOperationExecutor.java:301)
	at org.gradle.internal.operations.DefaultBuildOperationExecutor$RunnableBuildOperationWorker.execute(DefaultBuildOperationExecutor.java:293)
	at org.gradle.internal.operations.DefaultBuildOperationExecutor.execute(DefaultBuildOperationExecutor.java:175)
	at org.gradle.internal.operations.DefaultBuildOperationExecutor.run(DefaultBuildOperationExecutor.java:91)
	at org.gradle.internal.operations.DelegatingBuildOperationExecutor.run(DelegatingBuildOperationExecutor.java:31)
[... cut ...]
    implementation "com.android.support:cardview-v7:${supportLibraryVersion}"
    implementation "com.android.support:recyclerview-v7:${supportLibraryVersion}"
    implementation "com.android.support:support-v4:${supportLibraryVersion}"
    implementation "com.android.support:design:${supportLibraryVersion}"
    implementation "com.android.support:customtabs:${supportLibraryVersion}"
    implementation 'com.android.support:multidex:1.0.3'
    implementation 'com.android.support.constraint:constraint-layout:1.1.3'
`

What to do?

vhontar, 2019-05-08:
Duplicate classes is became visible after adding
implementation "org.jetbrains.kotlin:kotlin-stdlib:1.3.30"
before, there are problems which was described above.

avianey, 2019-05-08:
Your problem is not related...
Might come from somewhere else,
log indicates that you mix "androidx" with "android.support" which is not good
You should switch to androidx now
What is your dependency graph ?

sduqlsc, 2019-06-18:
same problem here, i fixed it through AS menu- tools->kotlin->configure kotiln in Project..

Hope there would be version written in `Java` and `Android Support Library`