RedApparat/Fotoapparat #375: Front camera : disabling mirror effect

root-ansh, 2019-06-04:
I used the following code to be able to open  and save a front cam picture, but the resultant picture I receive is mirrored ( preview's left becomes saved picture's right and vice versa)  

- code:  
```java   

    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_simple_cam);

        fotoapparat = getFotoApparatObj();

        fabCapture = findViewById(R.id.fab_clickpic);
        fabCapture.setOnClickListener(v -> {
            fabCapture.setEnabled(false);
            capturePicture();
        });

    }
    private void capturePicture() {
        String timeStamp = new SimpleDateFormat("yyyyMMdd_HHmmss", Locale.getDefault()).format(new Date());
        String imageFileName = "PIC_" + timeStamp + ".jpg";
        File tmpFile = new File(getExternalFilesDir(Environment.DIRECTORY_PICTURES), imageFileName);
        fotoapparat.takePicture().saveToFile(tmpFile).whenDone(unit -> onFileSuccesfullySaved(tmpFile));
    }
    private void onFileSuccesfullySaved(File tmpFile) {
        fabCapture.setEnabled(true);
        Uri tempFileUri = Uri.fromFile(tmpFile);
        Intent outputIntent = new Intent();

        if (tempFileUri != null) {
            setResult(RESULT_OK, outputIntent);
            outputIntent.putExtra(MainActivity.KEY_URI_STRING, tempFileUri.toString());
        } else {
            setResult(RESULT_CANCELED, outputIntent);
        }
        finish();

    }
    private Fotoapparat getFotoApparatObj() {
        Context ctx = this;
        CameraView cv = findViewById(R.id.camera_view);
        return Fotoapparat
                .with(ctx)
                .into(cv)
                .previewScaleType(ScaleType.CenterCrop)
                .photoResolution(ResolutionSelectorsKt.highestResolution())
                .lensPosition(LensPositionSelectorsKt.front()) 

                .focusMode(SelectorsKt.firstAvailable(
                        FocusModeSelectorsKt.continuousFocusPicture(),
                        FocusModeSelectorsKt.autoFocus(),        
                        FocusModeSelectorsKt.fixed()      
                ))
                .flash(SelectorsKt.firstAvailable(
                        FlashSelectorsKt.autoRedEye(),
                        FlashSelectorsKt.autoFlash(),
                        FlashSelectorsKt.torch()
                ))
                .build();
    }
}
```
- Sample gif : https://imgur.com/JEOoHOy   
- **FA version:**  2.7
- **Devices/APIs affected**:  api 21-28 (i use `minSDK` as 21 and `targetSDK` as 28)

root-ansh, 2019-06-05:
Also, i didn't wanted to open a new request for this, but we can't toggle/control HDR  , right?

AndyAls, 2019-08-07:
Have you solved this problem?

root-ansh, 2019-08-07:
not yet

Bridouille, 2019-09-07:
Hi, I ran into the same issue. For now, this is how I fixed it, applying a matrix transformation to the Bitmap:

```
fun Bitmap.flipAndRotate(degrees: Float): Bitmap {
    val matrix = Matrix().apply {
        preScale(1f, -1f)
        postRotate(degrees)
    }
    return Bitmap.createBitmap(this, 0, 0, width, height, matrix, true)
}
```

The `preScale(1f, -1f)` did the trick.

closed 2019-09-29 by root-ansh