RedApparat/Fotoapparat #379: If have two back camera device,how to choice which one to use?

DarkHorse-CN, 2019-06-13:
If have two back camera device,how to choice which one to use? 
[image]
[image]

erikbotta, 2019-08-20:
I am interested too. Xiaomi MI8 with telecamera for ex. And zoom is not working for me too. Is this library alive project?

qingdo, 2019-11-12:
It seems like they only implemented Front, Back and External (not sure what is this). I am also interested in this problem. The last update is Jun 2, so this lib is alive but may not be active right now. I would like to implement this feature and create a pull request (as I need to use this feature recently).