RedApparat/Fotoapparat #380: When turn on front Camera and back Camera Exception:E/Camera: Error 100

guoweiwen, 2019-06-23:
When my device turn on  front Camera and back Camera. On Stop Method call camera stop.

Exception:
D/Camera-JNI: (tid:21491)[native_release] + context=0xe24af500 camera=0xe2269ee0 
06-20 17:33:52.182 21254-21491/com.test.face D/Camera-JNI: [native_release] context->getStrongCount(2) camera->getStrongCount(2) 
06-20 17:33:52.184 21254-21254/com.test.face I/CameraFramework: handleMessage: 1 cameraID: 1
06-20 17:33:52.184 21254-26279/com.test.face W/IMediaDeathNotifier: media server died
06-20 17:33:52.185 21254-21491/com.test.face D/Camera-JNI: (tid:21491)[release camera] - X context=0xe24af500 
06-20 17:33:52.185 21254-21491/com.test.face D/Camera-JNI: (tid:21491)[~MtkJNICameraContext] this:0xe24af500 
06-20 17:33:52.185 21254-21254/com.test.face E/CameraFramework: Error 100
06-20 17:33:52.187 21254-26279/com.test.face E/MediaPlayer: error (100, 0)
06-20 17:33:52.188 21254-23162/com.test.face W/CameraBase: Camera service died!
06-20 17:33:52.192 21254-21254/com.test.face D/MediaPlayer: handleMessage msg:(100, 100, 0)
06-20 17:33:52.192 21254-21254/com.test.face E/MediaPlayer: Error (100,0)

```java
Initialized with:
if (fotoapparat_rgb == null) {
            fotoapparat_rgb = Fotoapparat.with(this)
                    .lensPosition(back())
                    .into(cameraView_rgb)
                    .frameProcessor(frameProcessor_rgb)
                    .cameraErrorCallback(new CameraErrorListener() {
                        @Override
                        public void onError(CameraException e) {
                            e.printStackTrace();
                            Log.e(TAG,e.getMessage());
                        }
                    })
                    .logger(loggers(
                            logcat(),
                            fileLogger(this)
                    )).build();
            fotoapparat_rgb.start();
        }
       if (fotoapparat_ir == null) {
            fotoapparat_ir = Fotoapparat.with(getApplicationContext())
                    .lensPosition(front())
                    .into(cameraView_ir)
                    .frameProcessor(frameProcessor_ir)
                    .cameraErrorCallback(new CameraErrorListener() {
                        @Override
                        public void onError(CameraException e) {
                            e.printStackTrace();
                            Log.e(TAG,e.getMessage());
                        }
                    })
                    .logger(loggers(
                            logcat(),
                            fileLogger(this)
                    ))
                    .build();
            fotoapparat_ir.start();
        }

Stop:
        @Override
    protected void onStop() {
        super.onStop();
        if (fotoapparat_rgb != null){
                fotoapparat_rgb.stop();
                fotoapparat_rgb = null;
        }
        if (fotoapparat_ir != null){
                fotoapparat_ir.stop();
                fotoapparat_ir = null;
        }
    }
```

guoweiwen, 2019-06-23:
Fotoapparat can support to open two cameras at the same time?

aznj, 2019-10-14:
any update on this?