RedApparat/Fotoapparat #391: Proguard issue: io.fotoapparat.coroutines.AwaitBroadcastChannel: can't find referenced method 'boolean cancel()' in program class kotlinx.coroutines.CompletableDeferred

ZakTaccardi, 2019-11-14:
```
io.fotoapparat.coroutines.AwaitBroadcastChannel: can't find referenced method 'boolean cancel()' in program class kotlinx.coroutines.CompletableDeferred
```

The issue is `AwaitBroadcastChannel`

https://github.com/RedApparat/Fotoapparat/blob/09a551635c6da6569911de0ba94ec40b5339781a/fotoapparat/src/main/java/io/fotoapparat/coroutines/AwaitBroadcastChannel.kt#L34-L36

I would guess that copying the deprecation warning and providing the correct `cancel(..)` that accepts a `CancellationException` would fix the issue? Or maybe just specifying deprecation level?

See [`ConflatedBroadcastChannel`](https://github.com/Kotlin/kotlinx.coroutines/blob/9d1146fdc91177320fb22c455086e14da0be008d/kotlinx-coroutines-core/common/src/channels/ConflatedBroadcastChannel.kt#L204-L220)

ZakTaccardi, 2019-11-14:
Is the following the correct solution?

```
-dontwarn io.fotoapparat.coroutines.AwaitBroadcastChannel
```

eduardoyurencb, 2020-01-27:
I have the same issue.

micaelomota, 2020-03-20:
I have the same issue.

OGx09, 2021-06-24:
And I'm here for this same issue. Has anyone found a solution to this problem?