RedApparat/Fotoapparat #407: Kotlin 1.4 update breaks the library's critical functionality

miloszkleszczewski, 2020-08-24:
With Kotlin's 1.4 update the Fotoapparat's `build()` function crashes with error: 

```
2020-08-24 10:57:08.883 7440-7440  E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com., PID: 7440
    java.lang.NoClassDefFoundError: Failed resolution of: Lkotlin/coroutines/experimental/CoroutineContext$Element;
        at io.fotoapparat.Fotoapparat.<init>(Fotoapparat.kt:55)
        at io.fotoapparat.Fotoapparat.<init>(Fotoapparat.kt:48)
        at io.fotoapparat.FotoapparatBuilder.buildInternal(FotoapparatBuilder.kt:191)
        at io.fotoapparat.FotoapparatBuilder.build(FotoapparatBuilder.kt:180)

```
Example code that should allow to reproduce: 

```
    private Fotoapparat createCamera() {
        return Fotoapparat
                .with(getContext())
                .into(cameraView)
                .previewScaleType(ScaleType.CenterCrop)
                .lensPosition(back())
                .build();
    }
```

AndyAls, 2020-09-25:
I have the same problem. Have you solved this problem

miloszkleszczewski, 2020-10-02:
@AndyAls I'm positive it's not possible to resolve it without modifying the library itself

WhiteYaksha, 2020-10-05:
Same issue here

miloszkleszczewski, 2021-01-10:
Version 2.7.0 works correctly

closed 2021-01-10 by miloszkleszczewski