RedApparat/Fotoapparat #413: JCenter to be sunset on 2020-05-01

petedoyle, 2021-02-04:
Hi! Just a heads up- JCenter will no longer be available after 2020-05-01:
https://jfrog.com/blog/into-the-sunset-bintray-jcenter-gocenter-and-chartcenter/

Just a note in case it makes sense to publish releases somewhere new (Maven Central, etc.)

Best,
Pete

ildar2, 2021-04-02:
only one month remaining... any updates?
@dmitry-zaitsev @Diolor @ansman

ShivlalChavan, 2021-12-09:
@Diolor any updates for replacing jcenter ?

dmitry-zaitsev, 2021-12-10:
@ShivlalChavan thanks for the heads up, I have finally have more time to work on the library and am planning to migrate off jcenter this week

borysboyko, 2021-12-16:
This is great news - thank you for working on this!

albertRCell, 2022-01-13:
@dmitry-zaitsev any updates on this topic?

matteogabella, 2022-02-07:
try this 

    implementation group: 'io.fotoapparat', name: 'fotoapparat', version: '2.7.0'
    implementation "androidx.annotation:annotation:1.3.0"
    implementation "androidx.exifinterface:exifinterface:1.3.3"
    implementation "org.jetbrains.kotlin:kotlin-stdlib:1.6.10"

borysboyko, 2022-03-13:
@matteogabella do you mean that these changes can be applied + jcenter() can then be removed? If so, this appears to not be working as io.fotoapparat:fotoapparat:2.7.0 cannot be found by Gradle within google() and mavenCentral() repositories.

milhauscz, 2023-07-31:
This library should be declared deprecated. It was great, but hasn't been updated since 2019, uses the deprecated Camera1 API and jcenter is not available anymore. I recommend everyone migration to [CameraX](https://developer.android.com/training/camerax).