RedApparat/Fotoapparat #420: CameraView not working

SherazAli-67, 2022-02-25:
Hi guys,

I was working on project that implements Fotoapparat it worked properly earlier but now when I am back for testing, Cameraview is not showing camera to capture image but capturing button is visible.

Here is the code 
```java

new Handler().postDelayed(() -> {
            foto = Fotoapparat
                    .with(getApplicationContext())
                    .into(camera_view)           // view which will draw the camera preview
                    .previewScaleType(ScaleType.CenterCrop)  // we want the preview to fill the view
                    .photoResolution(ResolutionSelectorsKt.highestResolution())   // we want to have the biggest photo possible
                    .lensPosition(fromScreen.equals("profilePicture") ? front() : back())       // we want back camera
                    .focusMode(SelectorsKt.firstAvailable(  // (optional) use the first focus mode which is supported by device
                            FocusModeSelectorsKt.continuousFocusPicture(),
                            FocusModeSelectorsKt.autoFocus(),        // in case if continuous focus is not available on device, auto focus will be used
                            FocusModeSelectorsKt.fixed()             // if even auto focus is not available - fixed focus mode will be used
                        ))
                        .flash(SelectorsKt.firstAvailable(      // (optional) sFimilar to how it is done for focus mode, this time for flash
                                FlashSelectorsKt.autoRedEye(),
                                FlashSelectorsKt.autoFlash(),
                                FlashSelectorsKt.off()
                        ))
                        .logger(LoggersKt.loggers(            // (optional) we want to log camera events in 2 places at once
                                LoggersKt.logcat(),           // ... in logcat
                                LoggersKt.fileLogger(this)    // ... and to file
                        ))
                    .cameraErrorCallback(e -> {
                        e.printStackTrace();
                        Log.d("TAG", "Error: "+e.getMessage());
                        Toast.makeText(this, "Chip[)a: "+e.getMessage(), Toast.LENGTH_SHORT).show();
                    })
                    .build();
        }, 200);

```

#### How did you initialize FA?

```java
@Override
    protected void onStart() {
        super.onStart();
        //cameraKitView.onStart();
        if (foto != null) {
            foto.start();
        }
    }

    @Override
    protected void onResume() {
        super.onResume();
    }

    @Override
    protected void onPause() {
        super.onPause();
    }

    @Override
    protected void onStop() {
        super.onStop();
        if (foto != null) {
            foto.stop();
        }
    }
```

When I run application this is what I am getting

[image]

andresteves, 2023-05-29:
Same issue, have you managed to solve it?

JohnLopen, 2024-08-19:
The same issue on Google Pixel 6 Pro. Any solution to this one please?