RedApparat/Fotoapparat #424: `takePicture()` freezes on API 33.

RobinFarmer, 2022-10-24:
Please check if your issue exists.

Issues should only be posted after you have been able to reproduce them and confirm that they are either a missing functionality or a bug. 

#### What are you trying to achieve or the steps to reproduce?

When trying to run the sample app on a Google Pixel running on API 33 (Android 13, TIRAMISU) taking a photo just freezes the preview and it never actually completes.

In the Log I can see the following being logged:

```
E/Camera: Error 2
E/Camera: Error 1
```

#### How did you initialize FA?

Just using the sample app, the only I changed was the sdk version inside of the top level `build.gradle` to support api 33.

```
sdk    : [
    minimum: 30,
    target : 33
],
```

#### What was the result you received?

Preview freezes and no picture is taken successfully.

#### What did you expect?

Normal camera behavior where the picture is taken and the preview just stays active.

#### Context:

* *FA version*: 2.7.0
* *Devices/APIs affected*: I used the Pixel 6 with API version 33.
* *any other relevant information*: N/A

noxo, 2022-10-24:
good to know something like this is expected, does it work if target sdk is 32?

RobinFarmer, 2022-10-25:
@noxo Thanks for your message, what do you mean that this is expected?

I tried the following (All on emulators):

Left SDK target on 33 top level `build.gradle`.

Run on Emulator that has API 28 -> works fine.
Run on Emulator that has API 30 -> works fine.
Run on Emulator that has API 31 -> Stopped working with `E/Camera: Error 2`
Run on Emulator that has API 32 -> Does not work with error: `E/Camera: Error 2`
Run on Emulator that has API 33 -> Does not work with error: `E/Camera: Error 2`

noxo, 2022-10-25:
Thanks for info. I've been observing same behaviour on emulator, but then it works on real device.

It's expected, because nobody seems to maintain the project anymore, so I would look for something else if you are looking for camera lib.

RobinFarmer, 2022-11-03:
Thanks, I was just able to test on a real device on API 33 and can confirm it does not happen. So it seems to be a problem with emulators...

guyca, 2023-02-10:
I can confirm that taking pictures seems to work on a real device. I guess it's time to migrate away from this library. What are y'all migrating to?

RobinFarmer, 2023-02-10:
We switched to CameraX which has been a great experience!