RedApparat/Fotoapparat #425: Android Studio Electric Eel problems.

isabsent, 2023-01-25:
Android Studio Electric Eel | 2022.1.1
Build #AI-221.6008.13.2211.9477386, built on January 11, 2023

Upgrade Gradle version to 7.5
Version 7.5 is the minimum version of Gradle compatible with Android Gradle Plugin version 7.4.0.

distributionUrl=https\://services.gradle.org/distributions/gradle-7.4-bin.zip    --> distributionUrl=https\://services.gradle.org/distributions/gradle-7.5-bin.zip

classpath 'com.android.tools.build:gradle:7.3.1'    --> classpath 'com.android.tools.build:gradle:7.4.0'

    Execution failed for task ':app:mergeExtDexFreeFatRelease'.
    > Could not resolve all files for configuration ':app:freeFatReleaseRuntimeClasspath'.
       > Failed to transform fotoapparat-2.7.0.aar (io.fotoapparat:fotoapparat:2.7.0) to match attributes {artifactType=android-dex, asm-transformed-variant=NONE, dexing-enable-desugaring=true, dexing-enable-jacoco-instrumentation=false, dexing-is-debuggable=true, dexing-min-sdk=23, org.gradle.category=library, org.gradle.libraryelements=jar, org.gradle.status=release, org.gradle.usage=java-runtime}.
          > Execution failed for DexingWithClasspathTransform: /Users/lev/.gradle/caches/transforms-3/4bdf7b32a06e61d7e0384ffe236baf54/transformed/jetified-fotoapparat-2.7.0-runtime.jar.
             > Error while dexing.

po4yka, 2023-02-03:
Faced with the same problem:

## Android Studio Information:
Android Studio Electric Eel | 2022.1.1 Patch 1
Build #AI-221.6008.13.2211.9514443, built on January 21, 2023

## Reproduction conditions
I have Gradle 8.0-rc-2 and everything works great:
`distributionUrl=https\://services.gradle.org/distributions/gradle-8.0-rc-2-all.zip`

But when I try to upgrade the AGP (`com.android.tools.build:gradle`) version from 7.3.1 to 7.4.0 during build sync I get the following error:

```
Execution failed for task ':app:mergeExtDexProjectNameDebug'.
> Could not resolve all files for configuration ':app:projectNameDebugRuntimeClasspath'.
   > Failed to transform fotoapparat-2.7.0.aar (io.fotoapparat:fotoapparat:2.7.0) to match attributes {artifactType=android-dex, asm-transformed-variant=NONE, dexing-enable-desugaring=true, dexing-enable-jacoco-instrumentation=false, dexing-is-debuggable=true, dexing-min-sdk=24, org.gradle.category=library, org.gradle.libraryelements=jar, org.gradle.status=release, org.gradle.usage=java-runtime}.
      > Execution failed for DexingNoClasspathTransform: /Users/po4yka/.gradle/caches/transforms-3/cbb562f61d3111478112c8c28298c2ec/transformed/jetified-fotoapparat-2.7.0-runtime.jar.
         > Error while dexing.
```

Other information:
- minSdkVersion = 21
- targetSdkVersion = 33
- compileSdkVersion = 33

dsvoronin, 2023-02-20:
Have same issue while updating to AGP 7.4.x

Stacktrace goes deeper:

```
org.gradle.api.tasks.TaskExecutionException: Execution failed for task ':app_retail:mergeExtDexDebug'.
```

```
Caused by: org.gradle.api.internal.artifacts.transform.TransformException: Failed to transform fotoapparat-2.7.0.aar (io.fotoapparat:fotoapparat:2.7.0) to match attributes {artifactType=android-dex, asm-transformed-variant=NONE, dexing-enable-desugaring=true, dexing-enable-jacoco-instrumentation=false, dexing-is-debuggable=true, dexing-min-sdk=24, org.gradle.category=library, org.gradle.libraryelements=jar, org.gradle.status=release, org.gradle.usage=java-runtime}.
```

```
Caused by: org.gradle.api.internal.artifacts.transform.TransformException: Execution failed for DexingNoClasspathTransform: /home/dsvoronin/.gradle/caches/transforms-3/343868255833a9def114c8a688a97412/transformed/fotoapparat-2.7.0-runtime.jar.
```

```
Caused by: com.android.builder.dexing.DexArchiveBuilderException: Error while dexing.
```

```
Caused by: com.android.tools.r8.CompilationFailedException: Compilation failed to complete
```

```
Caused by: com.android.tools.r8.internal.Fq: No VersionRequirement with the given id in the table

    at com.android.tools.r8.internal.LS.a(R8_4.0.48_320a4fe2564c68ad8fe4492fea65872bc8f51e21c4d46e3dbac43afcc38d90de:5679)
    at com.android.tools.r8.internal.LS.a(R8_4.0.48_320a4fe2564c68ad8fe4492fea65872bc8f51e21c4d46e3dbac43afcc38d90de:3962)
    at com.android.tools.r8.internal.LS.a(R8_4.0.48_320a4fe2564c68ad8fe4492fea65872bc8f51e21c4d46e3dbac43afcc38d90de:2968)
    at com.android.tools.r8.jetbrains.kotlinx.metadata.jvm.KotlinClassMetadata$SyntheticClass.c(R8_4.0.48_320a4fe2564c68ad8fe4492fea65872bc8f51e21c4d46e3dbac43afcc38d90de:5)
    at com.android.tools.r8.kotlin.S.a(R8_4.0.48_320a4fe2564c68ad8fe4492fea65872bc8f51e21c4d46e3dbac43afcc38d90de:1)
    at com.android.tools.r8.kotlin.j.a(R8_4.0.48_320a4fe2564c68ad8fe4492fea65872bc8f51e21c4d46e3dbac43afcc38d90de:33)
    at com.android.tools.r8.internal.OB.a(R8_4.0.48_320a4fe2564c68ad8fe4492fea65872bc8f51e21c4d46e3dbac43afcc38d90de:43)
    at com.android.tools.r8.utils.x.a(R8_4.0.48_320a4fe2564c68ad8fe4492fea65872bc8f51e21c4d46e3dbac43afcc38d90de:41)
    at com.android.tools.r8.utils.x.a(R8_4.0.48_320a4fe2564c68ad8fe4492fea65872bc8f51e21c4d46e3dbac43afcc38d90de:46)
    at com.android.tools.r8.utils.x.a(R8_4.0.48_320a4fe2564c68ad8fe4492fea65872bc8f51e21c4d46e3dbac43afcc38d90de:40)
    at com.google.common.util.concurrent.TrustedListenableFutureTask$TrustedFutureInterruptibleTask.runInterruptibly(TrustedListenableFutureTask.java:131)
    at com.google.common.util.concurrent.InterruptibleTask.run(InterruptibleTask.java:74)
    at com.google.common.util.concurrent.TrustedListenableFutureTask.run(TrustedListenableFutureTask.java:82)
    at com.google.common.util.concurrent.MoreExecutors$DirectExecutorService.execute(MoreExecutors.java:323)
    at com.google.common.util.concurrent.AbstractListeningExecutorService.submit(AbstractListeningExecutorService.java:69)
    at com.google.common.util.concurrent.AbstractListeningExecutorService.submit(AbstractListeningExecutorService.java:36)
    at com.android.tools.r8.utils.x.a(R8_4.0.48_320a4fe2564c68ad8fe4492fea65872bc8f51e21c4d46e3dbac43afcc38d90de:39)
    at com.android.tools.r8.utils.x.a(R8_4.0.48_320a4fe2564c68ad8fe4492fea65872bc8f51e21c4d46e3dbac43afcc38d90de:9)
    at com.android.tools.r8.utils.x.a(R8_4.0.48_320a4fe2564c68ad8fe4492fea65872bc8f51e21c4d46e3dbac43afcc38d90de:45)
    at com.android.tools.r8.utils.x.a(R8_4.0.48_320a4fe2564c68ad8fe4492fea65872bc8f51e21c4d46e3dbac43afcc38d90de:43)
    at com.android.tools.r8.utils.x.a(R8_4.0.48_320a4fe2564c68ad8fe4492fea65872bc8f51e21c4d46e3dbac43afcc38d90de:42)
    at com.android.tools.r8.internal.OB.a(R8_4.0.48_320a4fe2564c68ad8fe4492fea65872bc8f51e21c4d46e3dbac43afcc38d90de:35)
    at com.android.tools.r8.D8.b(R8_4.0.48_320a4fe2564c68ad8fe4492fea65872bc8f51e21c4d46e3dbac43afcc38d90de:9)
    at com.android.tools.r8.internal.H20.a(R8_4.0.48_320a4fe2564c68ad8fe4492fea65872bc8f51e21c4d46e3dbac43afcc38d90de:50)
    at com.android.tools.r8.D8.a(R8_4.0.48_320a4fe2564c68ad8fe4492fea65872bc8f51e21c4d46e3dbac43afcc38d90de:90)
    at com.android.tools.r8.D8.d(R8_4.0.48_320a4fe2564c68ad8fe4492fea65872bc8f51e21c4d46e3dbac43afcc38d90de:154)
    at com.android.tools.r8.D8.b(R8_4.0.48_320a4fe2564c68ad8fe4492fea65872bc8f51e21c4d46e3dbac43afcc38d90de:1)
    at com.android.tools.r8.internal.vk.a(R8_4.0.48_320a4fe2564c68ad8fe4492fea65872bc8f51e21c4d46e3dbac43afcc38d90de:24)
    at com.android.tools.r8.internal.vk.b(R8_4.0.48_320a4fe2564c68ad8fe4492fea65872bc8f51e21c4d46e3dbac43afcc38d90de:2)
    at com.android.tools.r8.D8.run(R8_4.0.48_320a4fe2564c68ad8fe4492fea65872bc8f51e21c4d46e3dbac43afcc38d90de:11)
    at com.android.builder.dexing.D8DexArchiveBuilder.convert(D8DexArchiveBuilder.java:120)
    at com.android.build.gradle.internal.dependency.BaseDexingTransform.process(DexingTransform.kt:294)
```

sgjesse, 2023-02-21:
As an FYI, this is the retraced stacktrace:
```
Caused by: com.android.tools.r8.jetbrains.kotlinx.metadata.InconsistentKotlinMetadataException: No VersionRequirement with the given id in the table

    at com.android.tools.r8.jetbrains.kotlinx.metadata.InconsistentKotlinMetadataException.<init>(InconsistentKotlinMetadataException.kt:8)
    at com.android.tools.r8.jetbrains.kotlinx.metadata.impl.ReadersKt.acceptVersionRequirementVisitor(readers.kt:424)
    at com.android.tools.r8.jetbrains.kotlinx.metadata.impl.ReadersKt.accept(readers.kt:251)
    at com.android.tools.r8.jetbrains.kotlinx.metadata.impl.ReadersKt.accept(readers.kt:201)
    at com.android.tools.r8.jetbrains.kotlinx.metadata.jvm.KotlinClassMetadata$SyntheticClass.accept(KotlinClassMetadata.kt:167)
    at com.android.tools.r8.jetbrains.kotlinx.metadata.jvm.KotlinClassMetadata$SyntheticClass.toKmLambda(KotlinClassMetadata.kt:145)
    at com.android.tools.r8.kotlin.KotlinSyntheticClassInfo.create(KotlinSyntheticClassInfo.java:46)
    at com.android.tools.r8.kotlin.KotlinClassMetadataReader.createKotlinInfo(KotlinClassMetadataReader.java:173)
    at com.android.tools.r8.kotlin.KotlinClassMetadataReader.getKotlinInfo(KotlinClassMetadataReader.java:55)
    at com.android.tools.r8.kotlin.KotlinMetadataRewriter.lambda$runForD8$2(KotlinMetadataRewriter.java:142)
    at com.android.tools.r8.utils.ThreadUtils.lambda$processItems$5(ThreadUtils.java:77)
    at com.android.tools.r8.utils.ThreadUtils.lambda$processItems$7(ThreadUtils.java:102)
    at com.android.tools.r8.utils.ThreadUtils.lambda$processItemsWithResults$3(ThreadUtils.java:69)
    at com.google.common.util.concurrent.TrustedListenableFutureTask$TrustedFutureInterruptibleTask.runInterruptibly(TrustedListenableFutureTask.java:131)
    at com.google.common.util.concurrent.InterruptibleTask.run(InterruptibleTask.java:74)
    at com.google.common.util.concurrent.TrustedListenableFutureTask.run(TrustedListenableFutureTask.java:82)
    at com.google.common.util.concurrent.MoreExecutors$DirectExecutorService.execute(MoreExecutors.java:323)
    at com.google.common.util.concurrent.AbstractListeningExecutorService.submit(AbstractListeningExecutorService.java:69)
    at com.google.common.util.concurrent.AbstractListeningExecutorService.submit(AbstractListeningExecutorService.java:36)
    at com.android.tools.r8.utils.ThreadUtils.lambda$processItemsWithResults$4(ThreadUtils.java:69)
    at com.android.tools.r8.utils.ThreadUtils.processItemsWithResults(ThreadUtils.java:66)
    at com.android.tools.r8.utils.ThreadUtils.processItems(ThreadUtils.java:99)
    at com.android.tools.r8.utils.ThreadUtils.processItems(ThreadUtils.java:85)
    at com.android.tools.r8.utils.ThreadUtils.processItems(ThreadUtils.java:77)
    at com.android.tools.r8.kotlin.KotlinMetadataRewriter.runForD8(KotlinMetadataRewriter.java:134)
    at com.android.tools.r8.D8.lambda$finalizeApplication$15(D8.java:401)
    at com.android.tools.r8.utils.Timing.time(Timing.java:373)
    at com.android.tools.r8.D8.finalizeApplication(D8.java:399)
    at com.android.tools.r8.D8.run(D8.java:341)
    at com.android.tools.r8.D8.lambda$run$1(D8.java:128)
    at com.android.tools.r8.utils.ExceptionUtils.withCompilationHandler(ExceptionUtils.java:80)
    at com.android.tools.r8.utils.ExceptionUtils.withD8CompilationHandler(ExceptionUtils.java:64)
    at com.android.tools.r8.D8.run(D8.java:125)
    at com.android.builder.dexing.D8DexArchiveBuilder.convert(D8DexArchiveBuilder.java:120)
    at com.android.build.gradle.internal.dependency.BaseDexingTransform.process(DexingTransform.kt:294)
```

mkj-gram, 2023-02-22:
Would it be possible for you to share a reproduction with us and use the existing bug here:
https://b.corp.google.com/issues/267334465

It looks like you are using an outdated kotlin tool-chain but without a reproduction it is hard to verify.

dsvoronin, 2023-02-22:
Sorry, no ready-to-use reproducer from me, maybe someone can help with that.

Problem is indeed in outdated kotlin. I did an update of kotlin(to 1.7.21) and AGP(7.4.1) and repacked fotoapparat lib.
Everything works fine now, issue is gone.
Only thing should be changed in code is maxBy/minBy and some more methods from stdlib in code. It need to be replaced with maxByOrNull, etc. You will see it in failing tests.

bahacan19, 2023-03-15:
> Sorry, no ready-to-use reproducer from me, maybe someone can help with that.
>
> Problem is indeed in outdated kotlin. I did an update of kotlin(to 1.7.21) and AGP(7.4.1) and repacked fotoapparat lib. [...]

Hi
Do you have a change list, commit or branch for this workaround?

chrisjenx, 2023-03-22:
Still testing by try: https://github.com/chrisjenx/Fotoapparat/releases/tag/2.8.0

GennadyDagman, 2024-06-19:
Are you going to publish this 2.8 release any time soon?