RedApparat/Fotoapparat #427: adding     implementation 'io.fotoapparat:fotoapparat:2.7.0' doenst work

Stefanistkuhl, 2023-03-10:
i added this line     implementation 'io.fotoapparat:fotoapparat:2.7.0' and this gives me this error FAILURE: Build completed with 9 failures.

1: Task failed with an exception.
-----------
* What went wrong:
Execution failed for task ':app:dataBindingMergeDependencyArtifactsDebug'.
> Could not resolve all files for configuration ':app:debugCompileClasspath'.
   > Could not find io.fotoapparat:fotoapparat:2.7.0.
     Required by:
         project :app

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
==============================================================================

2: Task failed with an exception.
-----------
* What went wrong:
Execution failed for task ':app:mergeDebugResources'.
> Could not resolve all files for configuration ':app:debugRuntimeClasspath'.
   > Could not find io.fotoapparat:fotoapparat:2.7.0.
     Searched in the following locations:
       - https://dl.google.com/dl/android/maven2/io/fotoapparat/fotoapparat/2.7.0/fotoapparat-2.7.0.pom
       - https://repo.maven.apache.org/maven2/io/fotoapparat/fotoapparat/2.7.0/fotoapparat-2.7.0.pom
     Required by:
         project :app

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
==============================================================================

3: Task failed with an exception.
-----------
* What went wrong:
Execution failed for task ':app:checkDebugAarMetadata'.
> Could not resolve all files for configuration ':app:debugRuntimeClasspath'.
   > Could not find io.fotoapparat:fotoapparat:2.7.0.
     Searched in the following locations:
       - https://dl.google.com/dl/android/maven2/io/fotoapparat/fotoapparat/2.7.0/fotoapparat-2.7.0.pom
       - https://repo.maven.apache.org/maven2/io/fotoapparat/fotoapparat/2.7.0/fotoapparat-2.7.0.pom
     Required by:
         project :app

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
==============================================================================

4: Task failed with an exception.
-----------
* What went wrong:
Execution failed for task ':app:mapDebugSourceSetPaths'.
> Could not resolve all files for configuration ':app:debugRuntimeClasspath'.
   > Could not find io.fotoapparat:fotoapparat:2.7.0.
     Searched in the following locations:
       - https://dl.google.com/dl/android/maven2/io/fotoapparat/fotoapparat/2.7.0/fotoapparat-2.7.0.pom
       - https://repo.maven.apache.org/maven2/io/fotoapparat/fotoapparat/2.7.0/fotoapparat-2.7.0.pom
     Required by:
         project :app

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
==============================================================================

5: Task failed with an exception.
-----------
* What went wrong:
Execution failed for task ':app:processDebugMainManifest'.
> Could not resolve all files for configuration ':app:debugRuntimeClasspath'.
   > Could not find io.fotoapparat:fotoapparat:2.7.0.
     Searched in the following locations:
       - https://dl.google.com/dl/android/maven2/io/fotoapparat/fotoapparat/2.7.0/fotoapparat-2.7.0.pom
       - https://repo.maven.apache.org/maven2/io/fotoapparat/fotoapparat/2.7.0/fotoapparat-2.7.0.pom
     Required by:
         project :app

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
==============================================================================

6: Task failed with an exception.
-----------
* What went wrong:
Execution failed for task ':app:mergeDebugAssets'.
> Could not resolve all files for configuration ':app:debugRuntimeClasspath'.
   > Could not find io.fotoapparat:fotoapparat:2.7.0.
     Searched in the following locations:
       - https://dl.google.com/dl/android/maven2/io/fotoapparat/fotoapparat/2.7.0/fotoapparat-2.7.0.pom
       - https://repo.maven.apache.org/maven2/io/fotoapparat/fotoapparat/2.7.0/fotoapparat-2.7.0.pom
     Required by:
         project :app

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
==============================================================================

7: Task failed with an exception.
-----------
* What went wrong:
Execution failed for task ':app:checkDebugDuplicateClasses'.
> Could not resolve all files for configuration ':app:debugRuntimeClasspath'.
   > Could not find io.fotoapparat:fotoapparat:2.7.0.
     Searched in the following locations:
       - https://dl.google.com/dl/android/maven2/io/fotoapparat/fotoapparat/2.7.0/fotoapparat-2.7.0.pom
       - https://repo.maven.apache.org/maven2/io/fotoapparat/fotoapparat/2.7.0/fotoapparat-2.7.0.pom
     Required by:
         project :app

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
==============================================================================

8: Task failed with an exception.
-----------
* What went wrong:
Execution failed for task ':app:desugarDebugFileDependencies'.
> Could not resolve all files for configuration ':app:debugRuntimeClasspath'.
   > Could not find io.fotoapparat:fotoapparat:2.7.0.
     Searched in the following locations:
       - https://dl.google.com/dl/android/maven2/io/fotoapparat/fotoapparat/2.7.0/fotoapparat-2.7.0.pom
       - https://repo.maven.apache.org/maven2/io/fotoapparat/fotoapparat/2.7.0/fotoapparat-2.7.0.pom
     Required by:
         project :app

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
==============================================================================

9: Task failed with an exception.
-----------
* What went wrong:
Execution failed for task ':app:mergeDebugNativeLibs'.
> Could not resolve all files for configuration ':app:debugRuntimeClasspath'.
   > Could not find io.fotoapparat:fotoapparat:2.7.0.
     Searched in the following locations:
       - https://dl.google.com/dl/android/maven2/io/fotoapparat/fotoapparat/2.7.0/fotoapparat-2.7.0.pom
       - https://repo.maven.apache.org/maven2/io/fotoapparat/fotoapparat/2.7.0/fotoapparat-2.7.0.pom
     Required by:
         project :app

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
==============================================================================

* Get more help at https://help.gradle.org

BUILD FAILED in 802ms
22 actionable tasks: 11 executed, 11 up-to-date

dodko159, 2023-03-28:
Version 2.7.0 seems to be hosted on JCenter, so you need to add JCenter to your repositories.
...
repositories {
...
jcenter()
}

sampene, 2024-05-23:
> Version 2.7.0 seems to be hosted on JCenter, so you need to add JCenter to your repositories. ... repositories { ... jcenter() }

centre() is deprecated

closed 2024-07-12 by Stefanistkuhl

harshpatel63, 2025-05-14:
mvn repositories do show SciJava Public. Hence, add the maven url to build.gradle (project level).

```
allprojects {
    repositories {
         ...
        maven {
            url "https://maven.scijava.org/content/repositories/public/"
        }
        ...
    }
}
```